openwrt/packages #8313: jq: 1.6 slow startup times

lukaszdrukasz, 2019-02-27:
jq-1.6 loads for over a second at my tp-link wdr-3600 before it starts doing any useful work. I'll be happy to provide any further information needed, only have no idea what it would be. Examples below show what happens with the simplest input/filter pair I could think of but the "waiting period" is consistent regardless of the complexity of the actual JSON to parse and filters to run.

**environment**: 

* TP-LINK WDR-3600
* openwrt built from source r9439

**symptoms**:

```
root@tplink:~# jq --version
jq-1.6
root@tplink:~# echo 0|time jq
0
real	0m 1.35s
user	0m 1.24s
sys	0m 0.01s

```
**expected behaviour (that of v1.5)**:

```
root@tplink:~# jq --version
jq-1.5
root@tplink:~# echo 0|time jq
0
real	0m 0.11s
user	0m 0.05s
sys	0m 0.00s

```

lukaszdrukasz, 2019-02-27:
I just realized I probably should have tagged this for the package maintainer: @ratkaj 

Sorry for any other possible breaches of etiquette; my first issue ever...

diizzyy, 2019-06-22:
Closing as it's not a bug/issue

closed 2019-06-22 by diizzyy