openwrt/packages #8315: [meta] Python in openwrt repo

ja-pa, 2019-02-27:
Hi,
I would like to add Ninja build system to [OpenWrt repo](https://github.com/openwrt/openwrt/). Since Ninja depends on python, it's also necessary to add python as well (or to me it sound reasonable [see commet](https://github.com/openwrt/openwrt/pull/1871#issuecomment-468007675) )

I would like to ask you for an opinion what would be the best approach? Move python package from packages feed to openwrt ? Or make a separate package in openwrt ? 
@commodo @jefferyto what do you think about that?

commodo, 2019-02-28:
so, in this case, the answer is "no, don't move Python to OpenWrt packages"
the `tools/` folder in OpenWrt [as I understand it] is standalone; things get build there, before the build system even takes a peek into packages; 
this design philosophy, sounds reasonable to be;

what you can do, is add a new `tools/python/Makefile` and feel free to copy bits from the Python Makefile from this repo [ the parts that deal with HostBuild particularly ]
maybe even use Python3 [since this seems to be the future]

this may complicate things for Python/Python3 in the packages feeds, but I guess we can find ways to handle this [if this happens]

but my personal opinion about ninja & meson [feel free to disagree with me on any of my opinions], is that "it looks nice, but i'm not convinced";
typically, anything that comes out-of-Google is for super-scale stuff, and bloated layers of software;
while ninja may seem fast on paper, it don't know what build improvement it may offer for OpenWrt build time [ 20% ? 30% ? 2x ? 5x ? improvement ] ?
i feel it has to be at least 2x and preferably 3x or higher to convince people to use it and to convert the current build-system to ninja/meson

having said that, what I recommend doing, is generating metrics; i.e.  current OpenWrt build is X, Ninja/Meson build time is X * Y [ Y being the factor of improvement ]
also, make sure that you do these metrics with reproduce-able builds ; i am not sure how close is OpenWrt to having all builds reproduce-able, but if you can say: we have Y improvement, and the binaries [for the target] are the same], then you start to convince people

to get there, what you could do is one of 2 things:
1. create your own OpenWrt fork that uses Ninja & Meson, and constantly produce nice metrics that are visible on the project web-page [ could just be a github.com readme ]; and constantly refine the process of using ninja on a server as well as on a local machine
2. add ninja in the packages feeds, and do the same things with metrics & refinement as the first point ;

this is my view on things, and should be considered as such :)

diizzyy, 2019-02-28:
I think final goal is to provide a toolchain that supports meson/ninja not convert the tree itself.

The issue is that projects are starting to provide only meson/ninja build files and I doubt we want to maintain our own set of build files and track upstream. There are a lot of data available about ninja and make and they all pretty much tells the same story. The main idea of switching from make to ninja (from my point of view) is to catch bugs before introducing framework for meson builds.

As for performance....
https://jpospisil.com/2014/03/16/replacing-make-with-ninja.html
https://david.rothlis.net/ninja-benchmark/
http://hamelot.io/programming/make-vs-ninja-performance-comparison/
...and so on

commodo, 2019-02-28:
> The issue is that projects are starting to provide only meson/ninja build files and I doubt we want to maintain our own set of build files and track upstream.

ah, I wasn't aware of this

makes sense to support Ninja/Meson in that case;
then, maybe start adding Ninja/Meson in the packages feeds, under [the devel folder](https://github.com/openwrt/packages/tree/master/devel) and start from there;

if people in core would consider it useful to migrate into OpenWrt core, then it will happen;
i don't know if there are packages in core that require Ninja/Meson [yet];

mhei, 2019-02-28:
> The issue is that projects are starting to provide only meson/ninja build files and I doubt we want to maintain our own set of build files and track upstream.

Yes, this is a/the real problem. I understand that upstream projects don't want to maintain different sets of build systems, so we need to add generic support for Ninja/Meson like cmake/autotools etc.
This need to be done in OpenWrt core, I don't see how this could be handled in packages.
The goal should be to have in package Makefiles just two line like

    ...
    include $(INCLUDE_DIR)/package.mk
    include $(INCLUDE_DIR)/meson.mk
    ...

or so to make it for package maintainers as easy as possible.

commodo, 2019-02-28:
> This need to be done in OpenWrt core, I don't see how this could be handled in packages.

I'm not the person to convince [or to argue with], whether this is to be done in core or packages :)
I only care about Python & some Python packages atm.
[ FWIW and just as an information: lang/python offers `python[3]-package.mk` which is something like `meson.mk` ; admittedly  `python[3]-package.mk` are not that great [for a number of reasons], but it's just an idea ]

But if done in core, then adding a `tools/python` folder is a good place to start [if python is a requirement].
And I am prepared to help out with any conflicts of Python/Python3 for theses packages [if  `tools/python` is accepted ] .

mhei, 2019-02-28:
@commodo: Ah, ok... now I got it how this could be done in packages. Thanks for clarification :+1: 
BTW, I'm using similiar stuff for PHP PECL packages.

Meanwhile I read https://github.com/openwrt/openwrt/pull/1871 so I think further discussion should take place there.

ja-pa, 2019-03-01:
Ok, I think that the best solution will be to integrate ninja and meson to packages/devel. So I'm closing this issue and related MR. Thanks for your suggestions.

closed 2019-03-01 by ja-pa