openwrt/packages #8321: ld: symbol(s) not found for architecture x86_64

JohnChancfz, 2019-03-01:
i-r2_lede/staging_dir/hostpkg/lib -lssl -lcrypto -Wl,-stack_size,1000000  -framework CoreFoundation -o python Programs/python.o libpython3.6.a -ldl  -framework CoreFoundation -lz
ld: warning: directory not found for option '-L/Volumes/OpenWrt/bpi-r2_lede/staging_dir/host/usr/lib'
ld: warning: directory not found for option '-L/Volumes/OpenWrt/bpi-r2_lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib'
ld: warning: directory not found for option '-L/Volumes/OpenWrt/bpi-r2_lede/staging_dir/host/usr/lib'
ld: warning: directory not found for option '-L/Volumes/OpenWrt/bpi-r2_lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/lib'
Undefined symbols for architecture x86_64:
  "_libintl_bindtextdomain", referenced from:
      _PyIntl_bindtextdomain in libpython3.6.a(_localemodule.o)
  "_libintl_dcgettext", referenced from:
      _PyIntl_dcgettext in libpython3.6.a(_localemodule.o)
  "_libintl_dgettext", referenced from:
      _PyIntl_dgettext in libpython3.6.a(_localemodule.o)
  "_libintl_gettext", referenced from:
      _PyIntl_gettext in libpython3.6.a(_localemodule.o)
  "_libintl_setlocale", referenced from:
      _PyLocale_setlocale in libpython3.6.a(_localemodule.o)
  "_libintl_textdomain", referenced from:
      _PyIntl_textdomain in libpython3.6.a(_localemodule.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[4]: *** [python] Error 1
make[3]: *** [/Volumes/OpenWrt/bpi-r2_lede/build_dir/hostpkg/Python-3.6.5/.built] Error 2
make[3]: Leaving directory `/Volumes/OpenWrt/bpi-r2_lede/feeds/packages/lang/python/python3'
make[2]: *** [package/feeds/packages/python/python3/host/compile] Error 2
make[2]: Leaving directory `/Volumes/OpenWrt/bpi-r2_lede'
make[1]: *** [/Volumes/OpenWrt/bpi-r2_lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWrt/bpi-r2_lede'
make: *** [world] Error 2

jefferyto, 2019-03-11:
Were you compiling Python 3 directly or building something that depends on Python?

neheb, 2019-07-02:
Closing as this looks like it comes from some LEDE fork. I don't know what changes have been made there.

closed 2019-07-02 by neheb