openwrt/packages #8329: net: ddns services url for domains.google.com needs http -> https

jwmaline, 2019-03-02:
# Issue template (remove lines from top till here)

Maintainer: @mark0n 
Environment: arch independent, head

Description:

domains.google.com erroring on current URL. Redirect to identical URL except http -> https

mark0n, 2019-03-03:
Could you please share the content of `/etc/config/ddns`? In particular I'm curious if it contains `option use_https '1'`.

jwmaline, 2019-03-14:
Apologies for making you read the manual to me. Though it was helpful, thanks!

closed 2019-03-14 by jwmaline

JFtico, 2019-03-14:
Actually, there is a chicken and egg problem here. The UI for DDNS does not show a field for 'Use HTTP Secure' if that setting is missing from the et/config/ddns entry.  So no way to set this unless one first uses UCI via SSH.
And furthermore, even if one does set via UCI, which DOES make it show up in the UI, there is a warning that HTTPS is not supported, even though it looks like it works (it used https to invoke google). See attached image below.

So the defect seems to be that creating a new DDNS config does not set the use_https property at all, one way or the other, therefore hiding it from GUI users.

I'd suggest the following changes:
For the LuCi component:
Add use_https to any new config created via clicking the 'add' button.
Remove the warning about https, seems spurious, as it works with google.

For the default config when the package is installed:
Add use_https options to both sample configs in the default /et/config/ddns file.

[image]

JFtico, 2019-03-16:
@mark0n Any feedback on the above note? Seems this one is worth re-opening.

mark0n, 2019-03-16:
> The UI for DDNS does not show a field for 'Use HTTP Secure' if that setting is missing from the et/config/ddns entry. So no way to set this unless one first uses UCI via SSH.

@JFtico I actually do see the "Use HTTP Secure" option when I create a new DDNS configuration.
[image]

Please double check that your `wget`/`curl` package supports HTTPS (e.g. ensure `wget` is installed rather than `wget-nossl`).