openwrt/packages #8330: nut-upsmon broken after upgrade from 18.06.1 to 18.06.2 - upsmon does not start anymore

petvoigt, 2019-03-03:
Maintainer: @cshoredaniel 
Environment: openwrt-18.06.2-x86-64-combined-ext4.img on an APU2C4

Description:
After the upgrade from version 18.06.1 to 18.06.2 nut-monitor does not start upsmon anymore. Instead I find the following error in my syslog:
```
# grep nut /var/log/messages
daemon.notice procd: /etc/rc.d/S51nut-monitor: cat: can't open '/var/etc/nut/nut.conf': No such file or directory
```
I am using nut-upsmon only, e.g. I am monitoring an UPS attached to another machine in my network and I have:
```
# cat /var/etc/nut/nut.conf 
MODE=netclient
```

petvoigt, 2019-03-04:
It is obviosuly a similar issue as https://github.com/openwrt/packages/issues/6644 - probably re-introduced. I get ```upsmon``` up by bypassing ```/etc/init.d/nut-monitor```:
```opkg install shadow ```
```chown -R nut:nut /var/etc/nut```
```su - nut```
```upsmon -D &```
Unfortunately,  I am not that familiar with the startup scripts that I can fix it on my own.

cshoredaniel, 2019-03-08:
AFIACT I've fixed this in master but received no feedback so missed 
window for backporting to 18.06.2.  Can you confirm that NUT from master 
does / does not fix the issue and report back?

petvoigt, 2019-03-08:
On Thu, 07 Mar 2019 18:57:25 -0800
Daniel Dickinson <notifications@github.com> wrote:

> AFIACT I've fixed this in master but received no feedback so missed 
> window for backporting to 18.06.2.  Can you confirm that NUT from
> master does / does not fix the issue and report back?
> 
> 
> 

I would like to test, if you can give me some more details on how to
easily test without having to re-install and staying on 18.06.02:

Which is the relevant commit and which files are affected. Is there a
way to copy these files in place on my machine?

cshoredaniel, 2019-03-20:
Sorry it's taking so long.  Technical difficulties combined with busy 
schedule.  Not forgotten, hopefully will explain in detail soon.

Regards,

Daniel

cshoredaniel, 2019-04-28:
@petvoigt 

Ok, technical difficulties resolved and systems (mostly) back online.  To use master in 18.06.2 do:

``
git clone git@github.com:/openwrt/packages
cd packages
git checkout openwrt-18.06
git checkout master -- net/nut

If you're using a src-link in feeds.conf you can just use this tree.  If you need it in git, then add commit and push to your local tree.

Er, I guess I forgot to ask, have you built OpenWrt before and/or with local trees vs. only official repos?  i.e. if you need more complete info, please let me know!

petvoigt, 2019-05-01:
On Sun, 28 Apr 2019 13:39:54 -0700
Daniel Dickinson <notifications@github.com> wrote:

> @petvoigt
>
> Ok, technical difficulties resolved and systems (mostly) back
> online.  To use master in 18.06.2 do:
>
> ``
> git clone git@github.com:/openwrt/packages
> cd packages
> git checkout openwrt-18.06 [...]

@cshoredaniel

I am using the official pre-built openwrt-18.06.2 and the
corresponding official package repo. Could you please give me a few
lines assistance on how to test your latest changes of nut packages.

I have installed following nut packages:
# opkg list-installed |grep nut
nut - 2.7.4-9
nut-common - 2.7.4-9
nut-upsc - 2.7.4-9
nut-upslog - 2.7.4-9
nut-upsmon - 2.7.4-9

Just to clarify: I am using nut monitoring only, my server is installed
on another machine in my network.

cshoredaniel, 2019-05-14:
So, the easiest way is to replace /etc/init.d/nut-monitor with (the current version, below):
```
#!/bin/sh /etc/rc.common

START=82
STOP=28
USE_PROCD=1
UPSMON_C=/var/etc/nut/upsmon.conf

nut_upsmon_conf() {
	local cfg="$1"
	local RUNAS val optval

	echo "# Config file automatically generated from UCI config" > "$UPSMON_C"

	config_get RUNAS "$cfg" runas "nutmon"
	[ -n "$RUNAS" ] && echo "RUN_AS_USER $RUNAS" >> "$UPSMON_C"
	runas="$RUNAS"

	config_get val "$cfg" minsupplies 1
	echo "MINSUPPLIES $val" >> "$UPSMON_C"

	config_get val "$cfg" shutdowncmd "/usr/sbin/nutshutdown"
	echo "SHUTDOWNCMD \"$val\"" >> "$UPSMON_C"

	config_get val "$cfg" notifycmd
	[ -n "$val" ] && echo "NOTIFYCMD \"$val\"" >> "$UPSMON_C"

	config_get val "$cfg" pollfreq 5
	echo "POLLFREQ $val" >> "$UPSMON_C"

	config_get val "$cfg" pollfreqalert 5
	echo "POLLFREQALERT $val" >> "$UPSMON_C"

	config_get val "$cfg" hostsync 15
	echo "HOSTSYNC $val" >> "$UPSMON_C"

	config_get val "$cfg" deadtime 15
	echo "DEADTIME $val" >> "$UPSMON_C"

	echo "POWERDOWNFLAG /var/run/killpower" >> "$UPSMON_C"

	config_get val "$cfg" onlinemsg
	[ -n "$val" ] && echo "NOTIFYMSG ONLINE \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" onbattmsg
	[ -n "$val" ] && echo "NOTIFYMSG ONBATT \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" lowbattmsg
	[ -n "$val" ] && echo "NOTIFYMSG LOWBATT \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" fsdmsg
	[ -n "$val" ] && echo "NOTIFYMSG FSD \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" commokmsg
	[ -n "$val" ] && echo "NOTIFYMSG COMMOK \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" commbadmsg
	[ -n "$val" ] && echo "NOTIFYMSG COMMBAD \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" shutdownmsg
	[ -n "$val" ] && echo "NOTIFYMSG SHUTDOWN \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" replbattmsg
	[ -n "$val" ] && echo "NOTIFYMSG REPLBATT \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" nocommmsg
	[ -n "$val" ] && echo "NOTIFYMSG NOCOMM \"$val\"" >> "$UPSMON_C"
	config_get val "$cfg" noparentmsg
	[ -n "$val" ] && echo "NOTIFYMSG NOPARENT \"$val\"" >> "$UPSMON_C"

	notifylist() {
		local value="$1"
		append optval "$value" "+"
	}

	setnotify() {
		local cfg="$1"
[... 141 lines cut ...]
	return 0
}

reload_service() {
	if pgrep upsmon >/dev/null 2>/dev/null; then
		local runas=nutmon
		build_config
		/usr/sbin/upsmon -c reload
	else
		restart
	fi
}

stop_service() {
	upsmon -c stop
}

service_triggers() {
	config_load nut_monitor
	interface_triggers "add_trigger"
	procd_add_reload_trigger "nut_monitor"
}
```
If you're only using nut-monitor that is all you should need.
Sorry for taking so long to reply.

cshoredaniel, 2019-05-14:
To clarify that all you should need if all you're using is the packages you listed, including nut-upsmon.

petvoigt, 2019-05-20:
On Mon, 13 May 2019 19:31:06 -0700
Daniel Dickinson <notifications@github.com> wrote:

> To clarify that all you should need if all you're using is the
> packages you listed, including nut-upsmon.
> 

Thanks for your brief instructions. I have replaced /etc/rc.common as
you suggested. Unfortunately, this completely broke at least network
connectivity so that I was forced to plug a serial console cable into
my machine.

I am not sure, if other things besides networking broke. I had to
replace rc.common immediately because I need the machine for my daily
work.

Did I understand you wrong in replacing /etc/rc.common solely?

cshoredaniel, 2019-05-26:
@petvoigt 
/etc/rc.common?  I must have mistyped before catching and edit (apparently too late), or something.  It's supposed to /etc/init.d/nut-monitor. Perhaps you saw the rc.common at the beginning of the script and replaced the wrong script?

petvoigt, 2019-05-26:
On Sat, 25 May 2019 17:54:04 -0700
Daniel Dickinson <notifications@github.com> wrote:

> @petvoigt 
> /etc/rc.common?  I must have mistyped before catching and edit
> (apparently too late), or something.  It's supposed
> to /etc/init.d/nut-monitor. Perhaps you saw the rc.common at the
> beginning of the script and replaced the wrong script?
> 

@cshoredaniel
Sorry, I am afraid it's my mistake, I mixed the file name with the
first line in /etc/init.d/nut-monitor.

I have just tested again. Unfortunately, nut-monitor still throws
errors and refuses to start:

Sun May 26 14:52:35 2019 daemon.notice procd: /etc/rc.d/S51nut-monitor: id: unknown user nutmon
Sun May 26 14:52:35 2019 daemon.notice procd: /etc/rc.d/S51nut-monitor: chgrp: unknown group
Sun May 26 14:52:35 2019 daemon.notice procd: /etc/rc.d/S51nut-monitor: id: unknown user nutmon
Sun May 26 14:52:35 2019 daemon.notice procd: /etc/rc.d/S51nut-monitor:
chgrp: unknown group

As stated in my second post, I have on my system user:group "nut:nut"
instead of user "nutmon". Either my auto-generated user:group is wrong
or it should be matched in your /etc/init.d/nut-monitor. Not sure, if
it is the only issue.

cshoredaniel, 2019-05-26:
@petvoigt Ah yes nutmon user:group was added since 18.06 branch...
Can you you set:

``option runas 'nut'``
in section ``config upsmon 'upsmon'``
in ``/etc/config/nut_monitor``?

That ought to runas the nut user instead of the new (not present on your system) 'nutmon' user.

petvoigt, 2019-05-26:
On Sun, 26 May 2019 07:37:14 -0700
Daniel Dickinson <notifications@github.com> wrote:

> @petvoigt Ah yes nutmon user:group was added since 18.06 branch...
> Can you you set:
> 
> ``option runas 'nut'``
> in section ``config upsmon 'upsmon'``
> in ``/etc/config/nut_monitor``?
> 
> That ought to runas the nut user instead of the new (not present on
> your system) 'nutmon' user.
> 

@cshoredaniel
Ah, that helped: Nut-monitor autostarted on boot and is still working.

Just one question concerning permissions: /var/etc/nut/ and all files
in here belong to "root:root" and not to "nut:nut". Is that right?

cshoredaniel, 2019-05-26:
The group on /var/etc/nut/nut.conf and /var/etc/nut/upsmon.conf in my dir is nutmon (would be nut in your case), but it's actually irrelevant because /var/etc/nut is root:root 750 AND for upsmon the parent process runs as root and forks a less privileged child for everything except shutting down the system (which requires root which is why it's done this way).
So the files may have nut group but I probably should just leave it root:root for a upsmon only situation.  Not going to rush on that though, as master is working for me, and now a close facsimile for you.

petvoigt, 2019-05-26:
Thank you for clarification and your support. I leave my system just as
it is right nown and I hope next OpenWRT upgrade just leaves nut-monitor
in a working situation ;-)

petvoigt, 2019-08-27:
On Mon, 26 Aug 2019 22:14:37 -0700
Daniel Dickinson <notifications@github.com> wrote:

> @petvoigt Apparently changes I thought were in the latest 18.06
> didn't make it -- I'm working on
> https://github.com/openwrt/packages/pull/9841 which syncs the
> relevant bits from master/19.07, but they won't be in 18.06 unless
> there is another maintenance release, AFAIK.
> 

@cshoredaniel, thanks for this information. I have recently upgraded to
18.06.4 and nut-monitor still appears to run correctly.

How can I quickly check, if 19.07 backports are in place on my 18.06.4
machine?

petvoigt, 2019-08-27:
On Tue, 27 Aug 2019 04:21:48 -0700
Daniel Dickinson <notifications@github.com> wrote:

> If you preserved /etc/init.d/nut-monitor across the sysupgrade (or
> put it back), then you'll have avoided the regression on nut-monitor.
> 

Hm, I definitely did not preserve or restore the
patched /etc/init.d/nut-monitor during the upgrade ;-)

I re-scanned the thread and I strongly suppose /etc/init.d/nut-monitor
is the version shipped with 18.06.4. As long as it is doing its job I
leave it untouched and exptect the 19.07 release ;-).

cshoredaniel, 2019-08-27:
If you want to be sure, copy /etc/init.d/nut-monitor and the script from 
this thread to a computer with diff and do a diff (diff -Nau original new).

HTH!  I'm glad it's working -- nut-monitor does have additional fixes, 
but they are for things like it failing when interfaces go up and down 
-- if that doesn't happen to you, then things will work just fine.  The 
script here was part of what is in 18.06.4 I think.

petvoigt, 2019-08-27:
@cshoredaniel, thanks. I did the diff and replaced the original 18.06.4
version of /etc/init.d/nut-monitor with the script here. UPS monitoring
is still working ;-)

neheb, 2020-01-22:
https://github.com/openwrt/packages/pull/9841 was merged. Closing.

closed 2020-01-22 by neheb