openwrt/packages #8334: package luci-app-commands will not work correctly

arricchito, 2019-03-03:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

arricchito, 2019-03-03:
package luci-app-commands frm 03.01.2019 git-19.060.76298-56f9052-1 will not work correctly. When entering a command not output will displayed. Older version works as desired.
I replaced in /usr/lib/lua/luci/view the 2 modules "commands.htm and commands_public.htm" with the older version.
.

diizzyy, 2019-03-28:
luci-app-commands is in https://github.com/openwrt/luci
Closing due to wrong repo

closed 2019-03-28 by diizzyy