openwrt/packages #8344: collectd - crashes because of interface change && no pid-file allocated

simonbytes, 2019-03-04:
# Issue template (remove lines from top till here)

Maintainer: Jo-Philipp Wich, Hannu Nyman (according to https://openwrt.org/packages/pkgdata/collectd)
Environment: OpenWrt version OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)

Description: collectd crashes if luci-app-wifischedule is used and activates wifi turn on / off via cron (all devices within br-wifistuff interface).
If I start collectd manually with the collectd -f command, following error shows up:

```
root@openwrt-hostname:~# collectd -f
plugin_load: plugin "netlink" successfully loaded.
plugin_load: plugin "iwinfo" successfully loaded.
plugin_load: plugin "memory" successfully loaded.
plugin_load: plugin "cpu" successfully loaded.
plugin_load: plugin "load" successfully loaded.
plugin_load: plugin "rrdtool" successfully loaded.
rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
plugin_load: plugin "processes" successfully loaded.
plugin_load: plugin "interface" successfully loaded.
plugin_load: plugin "dns" successfully loaded.
Initialization complete, entering read-loop.
dns plugin: Opening interface `br-wifistuff' failed: br-wifistuff: No such device exists (SIOCGIFHWADDR: No such device)
dns plugin: PCAP returned error Generic error.
Segmentation fault
```
Disabling DNS plugin restarts collectd successfully.
Probably surviving wifi-schedule actions. I'll keep an eye on it.

Furthermore, collectd does not create any PID file. 
My collectd config starts with:
```
BaseDir "/var/run/collectd"
Include "/etc/collectd/conf.d"
PIDFile "/var/run/collectd-run.pid"
PluginDir "/usr/lib/collectd"
TypesDB "/usr/share/collectd/types.db"
Interval 60
ReadThreads 2
```
Neither the standard PIDFile or my changed PIDFile string is used at all by collectd.
Even if collectd uses 10 Threads via htop:
```
28626 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.78 /usr/sbin/collectd -f
28627 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.06 /usr/sbin/collectd -f
28628 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.05 /usr/sbin/collectd -f
28629 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.07 /usr/sbin/collectd -f
28630 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.06 /usr/sbin/collectd -f
28631 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.06 /usr/sbin/collectd -f
28632 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.56 /usr/sbin/collectd -f
28633 root       25   5  4012  1992  1388 S  0.0  1.6  0:00.47 /usr/sbin/collectd -f
28614 root       25   5  4012  1992  1388 S  0.0  1.6  0:02.28 /usr/sbin/collectd -f
```
Probably this is because of the plugins I am using. (What is the config option 'ReadThreats' for?)

feckert, 2019-07-02:
@simonbytes does the Issues
* DNS
* PID

also happen on the current version 5.9.0 in openwrt master branch?

hnyman, 2019-07-02:
The DNS plugin has been problematic all the time, as it runs the a bit complicated data collection cycle using external pcap library.

See my debugging attempts a few years ago in https://github.com/collectd/collectd/issues/1227 where I inserted tons of debugging statements into collectd source, but never got into the root cause of the problem. The usage of pcap gets too complicated in multi-threaded callback-based environment.

My advice is to disable the DNS plugin, if it causes problems for you. Sorry.

(Your 18.06.1 is ancient, but I fear that there has been no major improvement of the basic operations of that plugin in the upstream collectd itself.)

feckert, 2019-08-27:
So I think we could close this issue, because we could not fix this as described by @hnyman

closed 2019-09-10 by feckert