openwrt/packages #8349: noddos: fails to build

nwf, 2019-03-06:
Maintainer: @StevenHessing 
Environment: OpenWrt HEAD

Description:

`noddos` fails to build with modern `ipset` library, after at least http://git.netfilter.org/ipset/commit/?id=55fdd96e331e920ee62bd816a572ac24f6dcd1ae.  The compiler spew is as follows:

```
make[5]: Entering directory '/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5'
[  1%] Building CXX object src/CMakeFiles/Identifier_test.dir/DeviceProfile.cxx.o
In file included from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.h:44:0,
                 from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.cxx:23:
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h: In member function 'bool Ipset::Exists()':
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:102:70: error: invalid conversion from'int (*)(const char*, ...)' to 'ipset_print_outfn {aka int (*)(ipset_session*, void*, const char*, ...)}' [-fpermissive]
             struct ipset_session *session = ipset_session_init(printf);
                                                                      ^
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:102:70: error: too few arguments to function 'ipset_session* ipset_session_init(ipset_print_outfn, void*)'
In file included from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:37:0,
                 from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.h:44,
                 from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.cxx:23:
/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/libipset/session.h:132:30: note: declared here
 extern struct ipset_session *ipset_session_init(ipset_print_outfn outfn,
                              ^~~~~~~~~~~~~~~~~~
In file included from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.h:44:0,
                 from /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/DeviceProfile.cxx:23:
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:109:17: error: 'ipset_envopt_parse' was not declared in this scope
             if (ipset_envopt_parse(session, IPSET_ENV_EXIST, NULL) < 0) {
                 ^~~~~~~~~~~~~~~~~~
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:109:17: note: suggested alternative: 'ipset_envopt_unset'
             if (ipset_envopt_parse(session, IPSET_ENV_EXIST, NULL) < 0) {
                 ^~~~~~~~~~~~~~~~~~
                 ipset_envopt_unset
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:116:95: error: 'ipset_session_error' was not declared in this scope
                 syslog (LOG_ERR, "Ipset: Can't commit for setname %s: %s", ipsetName.c_str(), ipset_session_error(session));
                                                                                               ^~~~~~~~~~~~~~~~~~~
/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5/src/Ipset.h:116:95: note: suggested alternative: 'ipset_session_report'
                 syslog (LOG_ERR, "Ipset: Can't commit for setname %s: %s", ipsetName.c_str(), ipset_session_error(session));
                                                                                               ^~~~~~~~~~~~~~~~~~~
                                                                                               ipset_session_report
src/CMakeFiles/Identifier_test.dir/build.make:75: recipe for target 'src/CMakeFiles/Identifier_test.dir/DeviceProfile.cxx.o' failed
make[5]: *** [src/CMakeFiles/Identifier_test.dir/DeviceProfile.cxx.o] Error 1
make[5]: Leaving directory '/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.5.5'
```

blueveryday, 2019-03-11:
yes, I have the same to you.
Have no diea with it.

cotequeiroz, 2019-03-12:
I have a tentative patch for this, but haven't run-tested it.
https://github.com/cotequeiroz/packages/commit/688462e8c7b4d49c231d1fec8aaea60b58bc42ba

blueveryday, 2019-03-13:
> I have a tentative patch for this, but haven't run-tested it.
> [cotequeiroz@688462e](https://github.com/cotequeiroz/packages/commit/688462e8c7b4d49c231d1fec8aaea60b58bc42ba)
> Despite the commit message, I'm not sure if I actually sumitted it to @StevenHessing.
> If anyone can test this, I'll open a PR here and upstream (I do have the patch for upstream master).

thank you sir.
it sucessed !
I had test your patches and makefile files,well done.

cotequeiroz, 2019-03-13:
Can you describe what tests have you run, so that I can properly document the PR?

blueveryday, 2019-03-13:
> Can you describe what tests have you run, so that I can properly document the PR?

scource code:https://github.com/coolsnowwolf/lede.git
errors:

> make[3]: Leaving directory '/home/h/Desktop/lede/package/libs/libnetfilter-conntrack'
time: package/libs/libnetfilter-conntrack/compile#0.33#0.13#0.64
make[3]: Entering directory '/home/h/Desktop/lede/feeds/packages/net/noddos'
touch /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.prepared_8ee5294a600a7ff1d022b10fd10c30f6_6664517399ebbbc92a37c5bb081b5c53_check
. /home/h/Desktop/lede/include/shell.sh; xzcat /home/h/Desktop/lede/dl/noddos-0.5.5.tar.xz | tar -C /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5
touch /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.prepared_8ee5294a600a7ff1d022b10fd10c30f6_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.configured_*
rm -f /home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/stamp/.noddos_installed
make[3]: [Makefile:84: /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 123 (ignored)
mkdir -p /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5
(cd /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=mips -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr;/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl -DCMAKE_PREFIX_PATH=/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl -DCMAKE_SKIP_RPATH=TRUE  /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5 )
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc
-- Check for working C compiler: /home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc -- works
-- Detecting C compiler ABI info
[... cut ...]
make[3]: Leaving directory '/home/h/Desktop/lede/feeds/packages/net/noddos'
time: package/feeds/packages/noddos/compile#4.07#0.53#12.82
make[2]: *** [package/Makefile:109: package/feeds/packages/noddos/compile] Error 2
make[2]: Leaving directory '/home/h/Desktop/lede'
make[1]: *** [package/Makefile:103: /home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/h/Desktop/lede'
make: *** [/home/h/Desktop/lede/include/toplevel.mk:218：world] 错误 2
h@h-PC:~/Desktop/lede$

blueveryday, 2019-03-13:
I take place for files . 
sucessed, 
log :

> make[3]: Leaving directory '/home/h/Desktop/lede/package/libs/libnetfilter-conntrack'
time: package/libs/libnetfilter-conntrack/compile#0.36#0.12#0.56
make[3]: Entering directory '/home/h/Desktop/lede/feeds/packages/net/noddos'
touch /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.prepared_5d76d9e07f5f6d8b769c49f19512d24d_6664517399ebbbc92a37c5bb081b5c53_check
. /home/h/Desktop/lede/include/shell.sh; xzcat /home/h/Desktop/lede/dl/noddos-0.5.5.tar.xz | tar -C /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5

Applying ./patches/010-openssl-1.1-fixes.patch using plaintext: 
patching file src/opensslfingerprint.cxx

Applying ./patches/020-Ipset.cxx-update-libipset-API-to-version-7.patch using plaintext: 
patching file src/Ipset.cxx
patching file src/Ipset.h
patch unexpectedly ends in middle of line
Hunk #2 succeeded at 124 with fuzz 1.
touch /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.prepared_5d76d9e07f5f6d8b769c49f19512d24d_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.configured_*
rm -f /home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/stamp/.noddos_installed
make[3]: [Makefile:84: /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 123 (ignored)
mkdir -p /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5
(cd /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=mips -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/usr;/home/h/Desktop/lede/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl -DCMAKE_PREFIX_PATH=/home/h/Desktop/lede/staging_dir/target-mips_24kc_musl -DCMAKE_SKIP_RPATH=TRUE  /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/noddos-0.5.5 )
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
[... cut ...]
SHELL= flock /home/h/Desktop/lede/tmp/.root-copy.flock -c 'cp -fpR /home/h/Desktop/lede/build_dir/target-mips_24kc_musl/luci-app-noddos/.pkgdir/luci-app-noddos/. /home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/root-ar71xx/'
touch /home/h/Desktop/lede/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp/.luci-app-noddos_installed
make[3]: Leaving directory '/home/h/Desktop/lede/feeds/luci/applications/luci-app-noddos'
time: package/feeds/luci/luci-app-noddos/compile#0.47#0.18#2.01
make[3]: Entering directory '/home/h/Desktop/lede/feeds/packages/net/ntpclient'

cotequeiroz, 2019-03-14:
I meant if you actually ran the app in your router.

blueveryday, 2019-03-14:
> I meant if you actually ran the app in your router.

I am on a business trip, I will test it this weekend.

blueveryday, 2019-03-15:
> I meant if you actually ran the app in your router.

I got some errors.....

`root@OpenWrt:/tmp# opkg install noddos*.ipk
Installing noddos (0.5.5-2) to root...
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for noddos:
 *      kernel (= 4.14.105-1-c2ffef75bd3568dbfe7d18082c40258a) *
 * opkg_install_cmd: Cannot install package noddos.
root@OpenWrt:/tmp# 
`

cotequeiroz, 2019-03-15:
Can you get ssh access?  
If so, what webserver are you using?  Check that it is running.  When using nginx, I sometimes end up with no `/etc/nginx/luci_nginx.conf`.  I haven't studied what happens.  I just remove `/overlay/upper/etc/nginx/luci_nginx*.conf` (this will restore `luci_nginx.conf` and `luci_nginx_ssl.conf` files).  In my case, they are marked as removed, so they do not show up in `/etc/nginx/`. Make sure you backup that directory first, just to be sure.  Then reboot.

blueveryday, 2019-03-15:
> Can you get ssh access?
> If so, what webserver are you using? Check that it is running. When using nginx, I sometimes end up with no `/etc/nginx/luci_nginx.conf`. I haven't studied what happens. I [...]

I remove the noddos and luci-app-noddos when compiling image.
flash it , It's run very well.

cotequeiroz, 2019-03-15:
So this does not fix it.  Thank you for testing it.

blueveryday, 2019-03-15:
> So this does not fix it. Thank you for testing it.

You are welcome.
If you have new idea for it, I will be glad to test it.
Thank you so much.

cotequeiroz, 2019-03-18:
I've tried to run it, no avail.  It appears that its website site (https://www.noddos.io/) is down.  When the app starts up, it gets the device profiles from that website, and hangs trying to download it.  @StevenHessing, what's the status of this?

StevenHessing, 2019-03-19:
I'll get the website that hosted the device profiles back up in one or two
days. Will need to look in to the ipset API changed

StevenHessing, 2019-03-19:
The web-site is back up:

root@WRT1200AC:~# /usr/bin/getnoddosdeviceprofiles;  if [ $? -gt 0 ]; then
service noddos reload; fi
Verified OK

cotequeiroz, 2019-03-19:
I've managed to test it some, and have posted noddos/noddos#53 with the ipset 7 changes to master.  I will probably open a new one adjusting `getnoddosdeviceprofiles` to account for `wget` being provided by `uclient-fetch`, which does not support the `--timestamping` option.

cotequeiroz, 2019-03-20:
> 
> 
> your mean is that noddos well done ?

It should work if you install `wget`, and the patch you've tried before.  The reason why it was locking up before was because https://noddos.io/ was down.  Now that's up again, it should work.  The reason I asked you to install wget as well is that noddos uses a flag that's not available with uclient-fetch's implementation of wget.

blueveryday, 2019-03-21:
> It should work if you install `wget`, and the patch you've tried before. The reason why it was locking up before was because https://noddos.io/ was down. Now that's up again, it should work. The [...]

thank you very much .
I will test it.

PR #8758 (2019-04-30): noddos: add libipset-7, uclient-fetch compatiblity
Maintainer: @StevenHessing 
Compile tested: mipsel-74kc, Asus RT-56U; arm, WRT3200ACM - master
Run tested: WRT3200ACM; basic testing done; service runs, but I'm not sure it does what's intended to do, but it does map the devices in the network.

Fixes #8349 
Description:
Old API compatibility was kept with a compatiblity shim.

Detect wget --timestamping support to make it compatible with uclient-fetch implementation of wget.

Patches were submitted upstream, but no response was received yet.
closed 2019-04-30 by neheb