openwrt/packages #8357: travelmate: Cannot connect to specific AP (leading space/special char probably)

sindastra, 2019-03-06:
Maintainer: @dibdot 
Environment: ar71xx, TP-Link WR902AC v1, OpenWrt 18.06.2

Description:

In Germany there's a store/chain called Real; They provide open WiFi APs for customers to use.
For some reason, travelmate works with everything I tested so far, except their WiFi.

Today while doing some more testing, I noticed the WiFi SSID has a leading space, namely: " real,-Wi-Fi" (or other special character, but seems to be a space, hard to tell when you just see a "blank" and not the hex code for the character 😉).

Anyway, I suspect the issue might be because of the leading "space"? It simply never connects to it.

Note: Their WiFi works fine when connecting directly with my smartphone (iOS 12), and yes, I see the leading "space" there too.

sindastra, 2019-03-06:
Additional info:

Just ran the `/etc/config/wireless` through `xxd`:

```
[...]
00000790: 2720 7265 616c 2c2d 5769 2d46 6927 0a09  ' real,-Wi-Fi'..
[...]
```

So on the wireless config file, it's stored as `0x20`, which in `ASCII` at least, is indeed a normal space.

dibdot, 2019-03-07:
Hi, nice catch. The leading space is no problem. The comma "," is the culprit ... I use this character currently as a file separator. I'll try to fix this ... stay tuned.

dibdot, 2019-03-07:
@sindastra see linked bugfix PR, would be nice if you could test that ... thanks!

sindastra, 2019-03-07:
I'll try it as soon as I can.

I guess I'll just replace the travelmate.sh with the new one. But out of curiosity, is there a more elegant/streamlined way of doing it? I'm asking to learn. (:

dibdot, 2019-03-07:
Yep, just download the raw file here at github (https://raw.githubusercontent.com/openwrt/packages/3e050a0a4a695a1c2991041eca60dcae4a5bc63e/net/travelmate/files/travelmate.sh) to your router and save it as /usr/bin/travelmate.sh.

sindastra, 2019-03-09:
And `chmod +x` ;)

I'm just in the store testing and it's working now! Just writing this using travelmate. :D

sindastra, 2019-03-09:
Update: Just saw that you updated the link. Now I don't know whether I'm testing the current or the previous version. I don't remember whether I updated travelmate a day or two ago. I can do a diff at home.

dibdot, 2019-03-09:
Cool, thanks for testing!
The other changes are already tested by me - I'll merge the PR now! ;-)

Thanks again for bringing this up!

closed 2019-03-09 by dibdot

sindastra, 2019-03-09:
Another update so you know which version I'm running right now, so you don't have to wait for me to get home and diff it. Sorry for the spamming:

```
# sha256sum /usr/bin/travelmate.sh 
b53f620c26cbaadc4f481b62fe5f9e93ee06d00e3471fcb83fd3388f5f2359ad
```

dibdot, 2019-03-09:
Happy shopping! ;-)

sindastra, 2019-03-09:
I was just wondering, does OpenWrt rebuild/update packages during the lifetime of a release so that travelmate will be updated sometime soon, or will I have to wait for the next point release (18.06.3)?

Of course I can just do the manual update and don't have to actually wait. I was just wondering about the release of the package version. (:

dibdot, 2019-03-10:
Travelmate 1.4.x will be part of OpenWrt 19.x. Current stable OpenWrt 18.06.x will stay with travelmate 1.2.x - I do not plan any backports.