openwrt/packages #8364: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash

birdie-github, 2019-03-08:
I'm running [OpenWRT 18.06.2](http://downloads.openwrt.org/releases/18.06.2/targets/ar71xx/generic/openwrt-18.06.2-ar71xx-generic-tl-wdr3600-v1-squashfs-sysupgrade.bin)

```
# opkg install unbound
Installing unbound (1.8.1-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/unbound_1.8.1-1_mips_24kc.ipk
Installing zlib (1.2.11-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/zlib_1.2.11-2_mips_24kc.ipk
Installing libopenssl (1.0.2q-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/libopenssl_1.0.2q-1_mips_24kc.ipk

Installing libunbound (1.8.1-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/libunbound_1.8.1-1_mips_24kc.ipk
Configuring zlib.
Configuring libopenssl.
Configuring libunbound.
Configuring unbound.
unbound: default root hints (built in rootservers.net)
unbound: default protocol configuration
unbound: default memory configuration
unbound: default recursion configuration
```

```
# opkg install libustream-openssl ca-certificates 
Installing libustream-openssl (2018-07-30-23a3f283-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/libustream-openssl_2018-07-30-23a3f283-1_mips_24kc.ipk
Installing ca-certificates (20180409-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/ca-certificates_20180409-2_all.ipk
Configuring libustream-openssl.
Configuring ca-certificates.

# opkg install ca-bundle
Installing ca-bundle (20180409-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/ca-bundle_20180409-2_all.ipk
Configuring ca-bundle.
```

```
# uclient-fetch https://www.internic.net/domain/named.cache
Downloading 'https://www.internic.net/domain/named.cache'
Connecting to 192.0.32.9:443
Writing to 'named.cache'
named.cache          100% |*******************************|  3315   0:00:00 ETA
Download completed (3315 bytes)

mv named.cache root.hints
```

Here's my unbound_ext.conf:

```
##############################################################################
# Extended user clauses added to the end of the UCI generated 'unbound.conf'
#
# Put your own forward:, view:, stub:, or remote-control: clauses here. This
# file is appended to the end of 'unbound.conf' with an include: statement.
# Notice that it is not part of the server: clause. Use 'unbound_srv.conf' to
# place custom option statements in the server: clause.
##############################################################################

server:
  tls-cert-bundle: /etc/ssl/cert.pem
  do-ip6: no

  root-hints: /etc/unbound/root.hints

forward-zone:
  name: "."
  forward-tls-upstream: yes
  forward-addr: 1.1.1.1@853#cloudflare-dns.com
```

```
Fri Mar  8 13:06:35 2019 daemon.info unbound: [2881:0] info: service stopped (unbound 1.8.1).
Fri Mar  8 13:06:35 2019 daemon.info unbound: [2881:0] info: server stats for thread 0: 1088 queries, 1088 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Fri Mar  8 13:06:35 2019 daemon.info unbound: [2881:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Fri Mar  8 13:07:06 2019 daemon.info procd: Instance unbound::unbound s in a crash loop 6 crashes, 0 seconds since last crash
```

All other files are unchanged.

simonbytes, 2019-03-10:
Read the wiki: https://openwrt.org/docs/guide-user/services/dns/unbound

unbound needs to be configured besides dnsmasq (which is the standard dns server) or on top of dnsmasq.

birdie-github, 2019-03-10:
Nothing in this manual says that any defaults must be changed, so that unbound could work.

It's crashing. That's a fact.

EricLuehrsen, 2019-03-13:
You would get "can't bind port" errors if their were a conflict with dnsmasq. You did get 1088 lookup's before the crash. Are there any automatic lookups you have scripted when you boot such that there is a common domain each time? A domain with a weirdly recursive CNAME or malformed DNSSEC-RR may expose a Unbound bug in its normal robustness to these things. (Don't forget to @ericluehrsen the maintainer next time.)

EricLuehrsen, 2019-03-13:
Maybe unrelated, but worth noting that Unbound 1.9.1 is being brought into 18.06 for OpenSSL 1.0.2 use improvements. #8395

EricLuehrsen, 2019-03-24:
@hnyman no response from @birdie-github. what would be a fair countdown to close?

birdie-github, 2019-03-24:
>  Are there any automatic lookups you have scripted when you boot such that there is a common domain each time?

No, nothing.

> Also what size of device do you have? Unbound allocates about 10MB RAM when it starts. (Don't forget to @EricLuehrsen the maintainer next time.)

```
root@OpenWrt:~# free
             total       used       free     shared    buffers     cached
Mem:        125272      37628      87644       1300       2308      13192
-/+ buffers/cache:      22128     103144
Swap:            0          0          0
```

Looks like I've got lots of free RAM, in fact over 85MB.

EricLuehrsen, 2019-03-29:
In your log file Unbound did run and was stopped. It spit out the first two lines of closing session statistics. It was restarted and crashed on the restart. I must assume it was an interface up event. I have had this problem long ago on weaker routers. Unbound cannot close shop fast enough to be out of the way of the next session. I added a delay for this in init scripts a long time ago. Maybe you have a older device or a lot of other packages? Maybe the sleep isn't long enough for your use case. Maybe the procd ifup delay is not sufficient, if you have lots of packages dependent on ifup. You can tweak these and see if it solves your problem. Also have you tried 1.9.1-1? It uses openssl a bit cleaner and this may help some execution bottlenecks. 
**/etc/init.d/unbound**
```
stop_service() {
  ... ...
  # Wait! on restart Unbound may take time writing closure stats to syslog
  pidof $PROG && sleep 1
}

service_triggers() {
  ... ...
  PROCD_RELOAD_DELAY=2000
  ... ...
}
```

birdie-github, 2019-03-31:
I launch unbound manually and it crashes, so timings/device speed/etc. most likely have nothing to do with this issue.

EricLuehrsen, 2019-03-31:
I really don't know how to repeat your problem nor do I have reports of a similar problem on 17.01, 18.06, or master. I respect that people have good reason not to publish too much about their configuration, but I will need more that you are willing to share, because this does not seem to be a generic issue. I can suggest something to try to remove UCI from the loop and get more log information.
**/etc/config/unbound**
```
config unbound
    option manual_conf '1'
```
**/etc/unbound/unbound.conf**
```
    # verbosity number, 0 is least verbose. 1 is default.
    verbosity: 4
```

birdie-github, 2019-04-27:
I'll get back to this issue once a new release of OpenWRT is made available.

closed 2019-04-27 by birdie-github

mariarti, 2019-04-29:
I had same issue, and for me helped command:
# `unbound-checkconf `
unbound-checkconf: no errors in /etc/unbound/unbound.conf

birdie-github, 2019-06-02:
I've long switched to stubby - it's miles faster and more reliable.