openwrt/packages #8382: shorewall: update to latest 5.2.0.x release (at least)?

brianjmurrell, 2019-03-11:
Maintainer: @wvdakker
Environment: 18.06

Description: I wonder if an update of the shorewall* packages to at least the most recent 5.2.0.x release could be made to pick up the latest bug fixes there.

In particular 5.2.0.5 fixes some locking specifically for OpenWRT.

brianjmurrell, 2019-03-11:
I'd push a PR my self, but the 18.06.2 SDK seems to be acting up -- or more likely I'm just not remembering (or the steps have changed since I last used it) all of the magic incantations needed to get it to build select packages.

wvdakker, 2019-03-11:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/openwrt/p
> ackages","title":"openwrt/packages","subtitle":"GitHub
> repository","main_image_url":"
> in GitHub","url":"
> https://github.com/openwrt/packages"}},"updates":{"snippets":[{"icon":"PERSON","message":"@brianjmurrell
> in #8382: I'd push a PR my self, but the 18.06.2 SDK seems to be
> needed to get it to build select packages."}],"action":{"name":"View
> Issue","url":"
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

If you found a problem in the 5.2.0.x release can you submit an issue
on the version?It can be easier to track down the fixes/issues.Then I
will make a PR for the 18.06 branch.

wvdakker, 2019-03-11:
PR made for 5.2.0.5

diizzyy, 2019-03-14:
@brianjmurrell 
Please close this as it's resolved

closed 2019-03-14 by hnyman

brianjmurrell, 2019-03-15:
@hnyman Was just waiting for the actual packages to show up at https://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/ before closing this.

wvdakker, 2019-03-15:
They will be available when 18.03 is released. Until then you have to build the package yourself.

hnyman, 2019-03-15:
cc @jow- 

> > Was just waiting for the actual packages to show up at https://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/ before closing this.
> 
> They will be available when 18.03 is released. Until then you have to build the package yourself.

No, the updated packages should be built immediately by the 18.06 buildbot.

I think that the 18.06 buildbot is temporarily offline since Tuesday morning, likely due to some unplanned incident.  Maybe @jow-  knows more.

jow-, 2019-03-16:
@hnyman - right. The build cluster needs to be rebuilt due to some expired server sponsorship. Will hopefully done by Tuesday or Wednesday

brianjmurrell, 2019-03-19:
Looks like it must be back online.  The new shorewall packages are there along with what looks like a full rebuild of all of the packages.

Cheers!