openwrt/packages #8390: pcscd - Cannot open PC/SC drivers directory

piotrekcrash, 2019-03-12:
Installed: pcscd, pcsc-tools
Device Zoweetek Smart Card Reader (EMV)

When run pcscd in debug by:
```
pcscd --foreground --debug --apdu --color
```
Result is:
```
hotplug_libusb.c:141:HPReadBundleValues() Cannot open PC/SC drivers directory: /usr/lib/pcsc/drivers
```
Full log:
```
00000000 debuglog.c:289:DebugLogSetLevel() debug level=debug
00000139 debuglog.c:310:DebugLogSetCategory() Debug options: APDU
00000101 pcscdaemon.c:352:main() Force colored logs
00001043 configfile.l:285:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000175 configfile.l:322:DBGetReaderListDir() Skipping non regular file: .
00000086 configfile.l:322:DBGetReaderListDir() Skipping non regular file: ..
00000098 pcscdaemon.c:662:main() pcsc-lite 1.8.23 daemon ready.
00000828 hotplug_libusb.c:141:HPReadBundleValues() Cannot open PC/SC drivers directory: /usr/lib/pcsc/drivers
00000098 hotplug_libusb.c:142:HPReadBundleValues() Disabling USB support for pcscd.
```

piotrekcrash, 2019-03-12:
Not installed "ccid" package. Install fix problem.
Sorry.

closed 2019-03-12 by piotrekcrash

dangowrt, 2019-03-12:
I was about to ask which driver are you trying to use...