openwrt/packages #8392: python-various: new packages have serious make config errors

EricLuehrsen, 2019-03-13:
:/home/github/openwrt-wrt-3200$ make menuconfig 
WARNING: Makefile 'package/feeds/packages/postgresql/Makefile' has a build dependency on 'zlib/host', which does not exist
tmp/.config-package.in:117658:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:117658:	symbol PACKAGE_seafile-seahub depends on PACKAGE_python-simplejson
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:27385:	symbol PACKAGE_python-simplejson is selected by PACKAGE_seafile-seahub
tmp/.config-package.in:4834:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:4834:	symbol PACKAGE_python3-requests is selected by PACKAGE_python3-influxdb
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29061:	symbol PACKAGE_python3-influxdb depends on PACKAGE_python3-requests
tmp/.config-package.in:29989:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29989:	symbol PACKAGE_python3-pytz is selected by PACKAGE_python3-pytz
tmp/.config-package.in:28528:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:28528:	symbol PACKAGE_python3-dateutil is selected by PACKAGE_python3-dateutil
tmp/.config-package.in:28206:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:28206:	symbol PACKAGE_python3-click-log is selected by PACKAGE_bigclown-control-tool
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:120084:	symbol PACKAGE_bigclown-control-tool depends on PACKAGE_python3-click-log
tmp/.config-package.in:30163:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:30163:	symbol PACKAGE_python3-six is selected by PACKAGE_python3-six
tmp/.config-package.in:29153:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29153:	symbol PACKAGE_python3-jsonpath-ng depends on PACKAGE_python3-ply
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29664:	symbol PACKAGE_python3-ply is selected by PACKAGE_python3-jsonpath-ng

@jefferyto, @commodo, @neheb

commodo, 2019-03-13:
Will take a look.

Thanks
Alex

cotequeiroz, 2019-03-13:
Here's a clue, from `lang/python/README`:

> * consider adding conditional DEPENDS for each variant ; so for each Python[3] package add `+PACKAGE_python-lxml:<dep>` as seen in the above example ; the reason for this is build-time reduction ; if you want to build Python3 only packages, this won't build Python & Python packages + dependencies ; this is a known functionality of OpenWrt build deps
>  * **there is an exception to the above consideration: if adding `+PACKAGE_python-lxml` conditional deps creates circular dependencies [for some weird reason], then this can be omitted**

I have started looking into what needs to be changed, but haven't got a list yet.

cotequeiroz, 2019-03-13:
Got a list, will open a PR now.

closed 2019-03-15 by hnyman