openwrt/packages #8408: MAC-Telnet-Server: Identify platform as "OpenWRT"

lars18th, 2019-03-13:
Hi,

In the current package "MAC-Telnet"
https://github.com/openwrt/packages/tree/master/net/mac-telnet
when enabling the server mode, the platform announced is `Linux` instead of `OpenWRT`.

I recommand to patch de file "config.h" in the original sources to change the plaform:
https://github.com/jow-/MAC-Telnet/blob/master/config.h#L45

This will be useful when you mix different devices in the network.
Regards.

lars18th, 2019-03-22:
Hi,

With this simple patch the work is done:
```
#elif defined(linux) || defined(__linux__)
-#define PLATFORM_NAME "Linux"
+#define PLATFORM_NAME "OpenWRT"
```

I hope someone update the package with this change!

diizzyy, 2019-03-28:
Please do a proper pull request if you think this is worth integrating.

closed 2019-03-28 by diizzyy