openwrt/packages #8419: Toolchain compilation error

fuqiang03, 2019-03-16:
Description:
Ubuntu 18.06 
gcc8.4
glibc2.27
Router Model:WRT1900acs
[image]
[image]
[image]

fuqiang03, 2019-03-16:
[image]

diizzyy, 2019-03-16:
Wrong repo...

Known bug on ARM
https://github.com/openwrt/openwrt/pull/1203 (HACK)
https://github.com/openwrt/openwrt/pull/1913 (Most likely better/proper fix)

closed 2019-03-21 by fuqiang03