openwrt/packages #8455: openssh: Cannot add second port

dbischof90, 2019-03-19:
Maintainer: @tripolar
Environment: OpenWRT 18.06.2

Description:
When trying to add a second port by uncommenting `#Port 22` in `/etc/ssh/sshd_config` and also adding `Port 222` to the file to add a second port, I get the message
```
sh: 222: unknown operand
```
after I tried to 

```
service sshd reload
```

The same works in my Arch Linux environment as is often quoted online in forums.

tripolar, 2019-03-20:
report this bug upstream this is not openwrt related

jow-, 2019-03-20:
@tripolar - actually it is OpenWrt related. Problem lies in line 24 of the init script: https://github.com/openwrt/packages/blob/master/net/openssh/files/sshd.init#L24

To fix this, the `-z $lport` expression should probably get quoted, and the grep command above amended with an `head -n 1` or so to only use the first port line.

tripolar, 2019-03-21:
fixed in https://github.com/openwrt/packages/commit/ba72d8d13944efaa4d9a8814d1b83ec52c3b39f0

closed 2019-03-21 by tripolar

jow-, 2019-03-22:
Thanks @tripolar !

dbischof90, 2019-03-22:
No, the issue is not fixed, just masked. 

The `awk` solution just takes the first argument and ignores all others. If I see that right then adding both ports as described initially just takes the first one and exits afterwards. So it still only sets only one port but doesn't crash if two are set.

tripolar, 2019-03-22:
wtf, feel free to fix it

tripolar, 2019-03-22:
feel free to fix it

tripolar, 2019-03-22:
it's fixed. mdns knows about the first listed port, it seems that procd only allows one port but feel free to fix this.

tripolar, 2019-03-22:
feel free to fix it but currently procd just handles one port. The last port written to it is the one ubus will report.

dbischof90, 2019-03-22:
Wasn't meant as criticism, just pointing it out in case someone finds this thread and has a similar issue. 

I have no experience with procd but should I manage it somehow, I'll let you know and we can do a PR. Of course it's appreciated if you're able to pull it off too.

tripolar, 2019-03-22:
i think you are on the wrong path:

> Avahi is a system which facilitates service discovery on a local network via the mDNS/DNS-SD protocol suite. This enables you to plug your laptop or computer into a network and instantly be able to view other people who you can chat with, find printers to print to or find files being shared. Compatible technology is found in Apple MacOS X (branded "Bonjour" and sometimes "Zeroconf").

You don't want to share all your ssh ports, only the one that is needed, to reach your service at least this is my impression.