openwrt/packages #8456: How to set up the Openconnect client on Openwrt?

lordarthas1995, 2019-03-20:
Something was wrong, check it please!

```
Sat Mar 16 09:20:32 2019 daemon.info openconnect[20669]: POST https://209.141.33.97:8030/
Sat Mar 16 09:20:32 2019 daemon.info openconnect[20669]: Connected to 209.141.33.97:8030
Sat Mar 16 09:20:32 2019 daemon.info openconnect[20669]: SSL negotiation with 209.141.33.97
Sat Mar 16 09:20:32 2019 daemon.info openconnect[20669]: Server certificate verify failed: signer not found
Sat Mar 16 09:20:32 2019 daemon.notice openconnect[20669]: Server SSL certificate didn't match: sha256:b2d2d2806b16de44b421d97af477766e492f92b8afdb262530969764ed6e1c78
Sat Mar 16 09:20:32 2019 daemon.notice openconnect[20669]: SSL connection failure: Error in the certificate.
Sat Mar 16 09:20:32 2019 daemon.notice openconnect[20669]: Failed to open HTTPS connection to 209.141.33.97
Sat Mar 16 09:20:32 2019 daemon.notice netifd: openconnect (20669): Failed to obtain WebVPN cookie
Sat Mar 16 09:20:32 2019 user.notice openconnect: bringing down openconnect
Sat Mar 16 09:20:32 2019 daemon.notice netifd: Interface 'openconnect' is now down
Sat Mar 16 09:20:32 2019 daemon.notice netifd: Interface 'openconnect' is setting up now
```

yousong, 2019-03-20:
Try adding `option serverhash <sha1-hash-of-the-crt>`

lordarthas1995, 2019-03-20:
@yousong How to type this command line in SSH screen

yousong, 2019-03-20:
You need to edit `/etc/config/network` and add that uci option to the `openconnect` section.  See the readme at https://github.com/openwrt/packages/tree/master/net/openconnect for an example.

lordarthas1995, 2019-03-20:
@yousong 没用，在接口那里一直显示0发送0接收
这是系统内核日志

Wed Mar 20 18:29:21 2019 daemon.notice netifd: Interface 'openconnect' is setting up now
Wed Mar 20 18:29:21 2019 user.notice openconnect: initializing...
Wed Mar 20 18:29:21 2019 user.notice openconnect: adding host dependency for 209.141.33.97 at openconnect
Wed Mar 20 18:29:21 2019 user.notice openconnect: adding host dependency for 209.141.33.97 at openconnect
Wed Mar 20 18:29:21 2019 user.notice openconnect: executing 'openconnect 209.141.33.97:8030 -i vpn-openconnect --non-inter --syslog --script /lib/netifd/vpnc-script  --servercert=b2d2d2806b16de44b421d97af477766e492f92b8afdb262530969764ed6e1c78 --no-system-trust --authgroup DEFAULT -u admin'
Wed Mar 20 18:29:21 2019 daemon.info openconnect[31802]: POST https://209.141.33.97:8030/
Wed Mar 20 18:29:21 2019 daemon.info openconnect[31802]: Connected to 209.141.33.97:8030
Wed Mar 20 18:29:21 2019 daemon.info openconnect[31802]: SSL negotiation with 209.141.33.97
Wed Mar 20 18:29:21 2019 daemon.info openconnect[31802]: Server certificate verify failed: signer not found
Wed Mar 20 18:29:21 2019 daemon.notice openconnect[31802]: Server SSL certificate didn't match: sha256:b2d2d2806b16de44b421d97af477766e492f92b8afdb262530969764ed6e1c78
Wed Mar 20 18:29:21 2019 daemon.notice openconnect[31802]: SSL connection failure: Error in the certificate.
Wed Mar 20 18:29:21 2019 daemon.notice openconnect[31802]: Failed to open HTTPS connection to 209.141.33.97
Wed Mar 20 18:29:21 2019 daemon.notice netifd: openconnect (31802): Failed to obtain WebVPN cookie
Wed Mar 20 18:29:21 2019 user.notice openconnect: bringing down openconnect
Wed Mar 20 18:29:21 2019 daemon.notice netifd: Interface 'openconnect' is now down

yousong, 2019-03-20:
Use `option serverhash sha256:b2d2d2806b16de44b421d97af477766e492f92b8afdb262530969764ed6e1c78` if you cannot get sha1 value.

lordarthas1995, 2019-03-20:
> Use `option serverhash sha256:b2d2d2806b16de44b421d97af477766e492f92b8afdb262530969764ed6e1c78` if you cannot get sha1 value.

It not work
https://i.postimg.cc/hP1YXd5x/TIM-20190320201304.png
https://i.postimg.cc/9QxNpjVQ/TIM-20190320201349.png

yousong, 2019-03-20:
please read the log and the reply here more carefully.

closed 2019-03-20 by yousong

lordarthas1995, 2019-03-20:
> please read the log and the reply here more carefully.

I do not understand why you close my issue?  Can you upload a video tutorial about how to set up Openconnect client on Openwrt? Please

And I don't know how to make "option serverhash sha256:xxx" work in /etc/config/network?