openwrt/packages #8457: yggdrasil: why is MIPS FPU emulator a dependency?

dengqf6, 2019-03-20:
@wfleurant @aparcar 
If I recall right, current version of Golang compiler already supports softfp.
Besides that, there is mips24kf, which has an FPU.

wfleurant, 2019-03-20:
The latest Golang compiler did not produce a usable mips32r2 binary for kernels lacking FPU emulation support.

If the case is 'mips' being too broad then we'll have to possibly exclude based on subtargets (if possible).

`DEPENDS:=$(GO_ARCH_DEPENDS) @IPV6 +kmod-tun +@(mips):KERNEL_MIPS_FPU_EMULATOR
`

Are you able to test on mips32K devices? I'll continue to look further into this... 

Is it possible these lantiq subtargets have hardware FPU? `xrx200 xway xway_legacy falcon ase`

jow-, 2019-03-20:
> +@(mips):KERNEL_MIPS_FPU_EMULATOR 

Please do not enable kernel features from packages! Doing so will bloat the kernel image for all targets when the package happens to be built along with the base system by the buildbots.

wfleurant, 2019-03-20:
right, there is better method im looking at lang/node/Makefile, `DEPENDS:=@(HAS_FPU||KERNEL_MIPS_FPU_EMULATOR)`

dengqf6, 2019-03-21:
There are already some Golang packages, which don't depend on FPU emulator in this repo, such as geth.
@jefferyto

commit e635f17b (2019-03-21): yggdrasil: use Golang buildroot appropriately, actually fix #8457

This supersedes #8465
No longer needs FPU emulator now
closed 2019-03-21 by aparcar