openwrt/packages #8463: libv4l not compiling with libstdcpp

neheb, 2019-03-20:
Maintainer: @thess 
Environment: mvebu, libstdcpp

Log:
```
make[8]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils/media-ctl'
  CC       libmediactl_la-libmediactl.lo
  CCLD     libmediactl.la
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
  CC       libv4l2subdev_la-libv4l2subdev.lo
  CCLD     libv4l2subdev.la
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
  CC       media-ctl.o
  CC       options.o
  CCLD     media-ctl
make[8]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils/media-ctl'
make[7]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils/media-ctl'
Making all in v4l2-compliance
make[7]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils/v4l2-compliance'
  CXX      v4l2_compliance-v4l2-compliance.o
  CXX      v4l2_compliance-v4l2-test-debug.o
  CXX      v4l2_compliance-v4l2-test-input-output.o
  CXX      v4l2_compliance-v4l2-test-controls.o
  CXX      v4l2_compliance-v4l2-test-io-config.o
  CXX      v4l2_compliance-v4l2-test-formats.o
  CXX      v4l2_compliance-v4l2-test-buffers.o
  CXX      v4l2_compliance-v4l2-test-codecs.o
  CXX      v4l2_compliance-v4l2-test-subdevs.o
  CXX      v4l2_compliance-v4l2-test-media.o
  CXX      v4l2_compliance-v4l2-test-colors.o
  CXX      v4l2_compliance-media-info.o
media-info.cpp: In function 'media_type mi_media_detect_type(const char*)':
media-info.cpp:68:28: error: variable 'std::ifstream uevent_file' has initializer but incomplete type
  std::ifstream uevent_file(uevent_path.c_str());
                            ^~~~~~~~~~~
media-info.cpp: In function 'std::__cxx11::string mi_media_get_device(__u32, __u32)':
media-info.cpp:109:28: error: variable 'std::ifstream uevent_file' has initializer but incomplete type
  std::ifstream uevent_file(uevent_path.c_str());
                            ^~~~~~~~~~~
Makefile:762: recipe for target 'v4l2_compliance-media-info.o' failed
make[7]: *** [v4l2_compliance-media-info.o] Error 1
make[7]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils/v4l2-compliance'
Makefile:467: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3/utils'
Makefile:580: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/v4l-utils-1.16.3'
Makefile:507: recipe for target 'all' failed
make[4]: *** [all] Error 2
```

rossbcan, 2019-08-20:
+feeds/packages/libs/libv4l/patches/021-fix-missing-includes.patch:
--- a/utils/v4l2-compliance/media-info.cpp	2019-08-20 05:53:45.729044576 -0400
+++ b/utils/v4l2-compliance/media-info.cpp	2019-08-20 05:54:26.667024830 -0400
@@ -19,7 +19,7 @@
 #include <dirent.h>

 #include <linux/media.h>
-
+#include <fstream>
 #include <iostream>
 #include <media-info.h>

--- a/utils/v4l2-ctl/media-info.cpp     2019-08-20 05:55:50.012001869 -0400
+++ b/utils/v4l2-ctl/media-info.cpp     2019-08-20 05:56:41.201746478 -0400
@@ -19,7 +19,7 @@
 #include <dirent.h>

 #include <linux/media.h>
-
+#include <fstream>
 #include <iostream>
 #include <media-info.h>

This reverses errors introduced by incorrect 020-add-missing-includes.patch

Anyone sorting this out for package update should reconcile and merge the two patches

thess, 2019-08-20:
Closed by #9785

PR #9785 (2019-08-21): libv4l: Update to 1.16.6. Fix missing includes patch
Maintainer: me
Compile tested: mvebu with libstdc++ and uclibc++
Run tested: N/A

Description:
Use latest released version 1.16.6
Update/add missing includes -- fixes #8463
closed 2019-08-20 by thess