openwrt/packages #8469: Request for commit access

BKPepe, 2019-03-21:
Hello guys,

according to [CONTRIBUTING.md](https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md), I'd like to request commit access to this repository if it is possible.  

There is said [since 2014](https://github.com/openwrt/packages/commit/c8d252426688b20c9d5685192cc24ef459776a60#diff-6a3371457528722a734f3c51d9238c13), I can request it in next pull request, which I don't think that's a good way. My opinion that for that should be created an issue like it was done in #4372 and #5486. First, I chose the second way, which was to create a ticket. I did that on [8th March](https://bugs.openwrt.org/index.php?do=details&task_id=2173), where I a little bit introduced myself, but so far no response. I thought that I will try to push it via #openwrt-devel, but first, let's try this way as someone did it. :-)

Currently, I'm co-maintainer of a package **youtube-dl**, where I'm really active and I don't want to bother @hnyman too often and BTW: He is doing a great job here. Then I take care as maintainer for packages: **python-dns**, **syslog-ng**, **python-requests**, **python-paho-mqtt**. 

CC: @diizzyy and @neheb for these two guys it could be quite interesting as well as they are really active and useful for reviews. Nobody else is doing it so often like them. 

Ping: @thess, @stintel, @champtar 

If you'd like to ask anything, I'd be happy to answer it.

Thank you and looking forward to hearing from you,
Josef

neheb, 2019-03-21:
ACK.

I'm burned out at this point.

jow-, 2019-03-22:
ACK from me as well for @BKPepe @neheb @diizzyy @feckert @cotequeiroz

diizzyy, 2019-03-22:
Fine with me

feckert, 2019-03-22:
thank you I accept gratefully :-)

cotequeiroz, 2019-03-22:
I’m honored to be included in the list, and accept the invitation. Thank you.

thess, 2019-03-23:
Hey @members 

I definitely agree with @jow here. We need more people to have commit access.
> ACK from me as well for @BKPepe @neheb @diizzyy @feckert @cotequeiroz

It's time to turn over the asylum to the inmates - at least those who are doing all the work. Any objections? If none these 5 will be granted write access later this week.

champtar, 2019-03-23:
@thess @jow- I also agree, let's bring the people doing the work in

dangowrt, 2019-03-23:
ACK and I also believe we should be welcoming everyone willing to step up.

hnyman, 2019-03-24:
Fine with me.

dibdot, 2019-03-24:
I'd be really happy to have you guys on the team.

poranje, 2019-03-26:
> Hey @members
>
> I definitely agree with @jow here. We need more people to have commit access.
>
> ACK from me as well for @BKPepe @neheb @diizzyy @feckert @cotequeiroz
>
> It's time to turn over the asylum [...]

No objections whatsoever, happy to see more responsible inmates, but addition of a rule to the Contributing Guidelines [0] to the effect that one will not commit ones own changes would be in order to ensure kind of four eyes principle, not out of distrust but just common sense.

[0] https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md#if-you-have-commit-access

Regards,
Paul

thess, 2019-03-26:
@poranje - So, make a PR for the change.
I'm not sure I agree - but this is not the thread to discuss it. My feeling about the responsibility of committers is: "If you break it, you fix it". We don't want merges just for the purpose of trying something.

thess, 2019-03-26:
OK folks - Invites sent to Josef, Daniel, Rosen, Florian and Eneas.

Welcome aboard!

Please make sure you are somewhat familiar with our guidelines for merging, in particular the section mentioned above:

https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md#if-you-have-commit-access

If you also want commit access to `Luci` - Just post a note in that repo.

closed 2019-03-26 by thess