openwrt/packages #8498: golang: failed to compile hello world on mips

feckert, 2019-03-25:
Maintainer: @jefferyto 
Environment: lantiq_xrx200/x86_64, APU3, master
Description:

I have added the hello world program for golang as package.
To get familiar with the go build system. But I run into a problem with MIPS!

On x64_64 the program compiles on my build system (x86) and also works on my target.
On MIPS the program does not even compile. I get the following errors during building

Any hint what I am doing wrong?

> Building targets
> errors
> internal/cpu
> math/bits
> unicode/utf8
> runtime/internal/sys
> internal/race
> unicode
> runtime/internal/math
> sync/atomic
> runtime/cgo
> internal/bytealg
> runtime/internal/atomic
> internal/testlog
> math
> runtime
> \# runtime/cgo
> gcc_mipsx.S: Assembler messages:
> gcc_mipsx.S:25: Error: invalid operands `sw $18,12($29)'
> gcc_mipsx.S:26: Error: invalid operands `sw $19,16($29)'
> gcc_mipsx.S:27: Error: invalid operands `sw $20,20($29)'
> gcc_mipsx.S:28: Error: invalid operands `sw $21,24($29)'
> gcc_mipsx.S:29: Error: invalid operands `sw $22,28($29)'
> gcc_mipsx.S:30: Error: invalid operands `sw $23,32($29)'
> gcc_mipsx.S:31: Error: invalid operands `sw $30,36($29)'
> gcc_mipsx.S:44: Error: invalid operands `jalr $20'
> gcc_mipsx.S:48: Error: invalid operands `lw $18,12($29)'
> gcc_mipsx.S:49: Error: invalid operands `lw $19,16($29)'
> gcc_mipsx.S:50: Error: invalid operands `lw $20,20($29)'
> gcc_mipsx.S:51: Error: invalid operands `lw $21,24($29)'
> gcc_mipsx.S:52: Error: invalid operands `lw $22,28($29)'
> gcc_mipsx.S:53: Error: invalid operands `lw $23,32($29)'
> gcc_mipsx.S:54: Error: invalid operands `lw $30,36($29)'
> gcc_mipsx.S:63: Error: invalid operands `lw $31,0($29)'
> strconv
> sync
> io
> reflect
> syscall
> internal/syscall/unix
> time
> sort
> internal/poll
> internal/fmtsort
> os
> fmt

OpenWrt Makefile
```
#
# Copyright (C) 2019 Florian Eckert
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=gohello
PKG_VERSION:=0.1
PKG_RELEASE:=1
PKG_LICENSE:=GPL-2.0+

PKG_BUILD_DEPENDS:=golang/host
PKG_BUILD_PARALLEL:=1

GO_PKG:=tdt.de/fe/gohello

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/../feeds/packages/lang/golang/golang-package.mk

define Package/gohello
$(call GoPackage/GoSubMenu)
  TITLE:=Hello world implementaion for Go
  DEPENDS:=$(GO_ARCH_DEPENDS)
  PKGARCH:=all
endef

define Package/gohello/description
Hello world implementation for Go
endef

define Package/gohello/install
	$(call GoPackage/Package/Install/Bin,$(1))
endef

$(eval $(call GoSrcPackage,gohello))
$(eval $(call BuildPackage,gohello))
```

Source for src/hello.go

```
package main

import "fmt"

func main() {
	fmt.Println("hello world")
}

```

jefferyto, 2019-03-26:
Can you post the complete build log? Also are you sure the error occurs when compiling your test program and not the Go compiler?

I suggest adding `PKG_USE_MIPS16:=0` as I believe there would be errors if it is left enabled.

I also suggest removing the `PKGARCH:=all` as that tells the build system that the package is architecture-independent (though I don't think this is causing your error).

Also, if you call `GoBinPackage` instead of `GoSrcPackage`, then the default install recipe will install the binary (so you don't need to manually define `Package/gohello/install`).

diizzyy, 2019-03-27:
MIPS isn't supported...
https://github.com/golang/go/wiki/MinimumRequirements

jefferyto, 2019-03-27:
@diizzyy I beg to differ: https://golang.org/doc/install/source#introduction

Also: I've done builds for MIPS.

diizzyy, 2019-03-27:
So it seem like the GitHub repo isn't up to date, my bad.

feckert, 2019-03-28:
I did all your changes you suggest but nothing fixed my problem. So what do you need or what can I do to get this fixed.
Does Prometheus builds on MIPS?

jefferyto, 2019-03-28:
@feckert Yes it does: https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/prometheus_2.8.0-1_mips_24kc.ipk

Can you post the build log?

feckert, 2019-03-29:
Hello @jefferyto thank you very much for your help.
I have pasted my compile output to https://pastebin.com/NQmZy7g2

jefferyto, 2019-03-30:
I'm able to reproduce the error using your Makefile and source file. When I add `PKG_USE_MIPS16:=0` to the Makefile, the error is gone.

Are you sure you have added `PKG_USE_MIPS16:=0`?

feckert, 2019-04-01:
This was my fault!
I recreated my complete build environment again and enabled PKG_USE_MIPS16:=0 on the my go package and it is now working!!

Thanks for help and time

closed 2019-04-01 by feckert

jefferyto, 2019-04-01:
Glad to hear it's working :laughing: