openwrt/packages #8500: czmq: Update package metadata

karlp, 2019-03-26:
@ja-pa  The newly added czmq is..

* Missing a project URL
* Missing a useful description.
* Missing a useful title  (the title doesn't have to be the same letter soup as the name!)
* uses a special char in the description that doesn't render in menuconfig

https://github.com/openwrt/packages/blob/82ead2596b334b00a95a461ebec73225f91e364d/libs/czmq/Makefile#L36

Please consider improving this.

jow-, 2019-03-26:
While revising the Makefile, also consider adding `ABI_VERSION:=4` (according to https://github.com/zeromq/czmq/blob/master/CMakeLists.txt#L385) and change the `libczmq.so*` pattern to `libczmq.so`**.**`*` to only copy versioned `.so` files and symlinks.

The unversioned symlink is only needed at link-time.

ja-pa, 2019-03-27:
@karlp @jow- Hi I created PR with changes related to this issue see https://github.com/openwrt/packages/pull/8507 Could you please review my changes? Thanks for comments.

closed 2019-03-27 by BKPepe