openwrt/packages #8529: acme: luci-app-acme requires luci-lib-ipkg

karlp, 2019-03-28:
Maintainer: @tohojo 
Environment: master luci git.

Description:
the acme-cbi uses the luci.model.ipkg lua module to check what webserver is installed.  This module is "deprecated" in current luci, and has been moved to an extra package.  The acme app either needs to move to another method, or make sure it's depending on that package.

tohojo, 2019-04-01:
Karl Palsson <notifications@github.com> writes:

> Maintainer: @tohojo
> Environment: master luci git.
>
> Description:
> the acme-cbi uses the luci.model.ipkg lua module to check what
> webserver is installed.  This module is "deprecated" in current luci, [...]

Ah, bugger. Any suggestions for what "another method" might be?

jow-, 2019-04-01:
Maybe a simple test for the actual executables? Like `nxio.fs.access("/usr/sbin/uhttpd")`

tohojo, 2019-04-01:
Jo-Philipp Wich <notifications@github.com> writes:

> Maybe a simple test for the actual executables? Like
> `nxio.fs.access("/usr/sbin/uhttpd")`

Sure, can do...

orangepizza, 2019-04-21:
but wouldn't it break with luci with nginx?

> 
> 
> Maybe a simple test for the actual executables? Like `nixio.fs.access("/usr/sbin/uhttpd")`

jow-, 2019-04-22:
> but wouldn't it break with luci with nginx?

No.

commit 279c09d0 (2019-05-23): ACME: Remove dependency on luci-lib-ipkg

Instead, use @jow-'s suggestion of just checking for the presence of the
executables to find the installed web servers.

Fixes #8529.
closed 2019-05-23 by tohojo