openwrt/packages #8533: miniupnpd: some ports only forwarded after 'enable additional logging' checked

rodibin-at-placemaildotonline, 2019-03-29:
# Issue template (remove lines from top till here)

**Maintainer:** @neheb 
**Environment:** 
Model
Linksys WRT1900ACS
Architecture
ARMv7 Processor rev 1 (v7l)
Firmware Version
OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d) 

**Description:**
I'm using transmission from Termux. I filed a bug report here: https://github.com/termux/termux-packages/issues/3588 maintained by @xeffyr 

upnpd on OpenWrt only works with transmission-daemon on termux when "enable additional logging" is checked

_Termux log_
```
[2019-03-28 22:26:26.682] Port Forwarding (UPnP) Found Internet Gateway Device "http://192.168.1.1:5000/" (/home/builder/.termux-build/transmission/src/libtransmission/upnp.c:>
[2019-03-28 22:26:26.682] Port Forwarding (UPnP) Local Address is "192.168.1.247" (/home/builder/.termux-build/transmission/src/libtransmission/upnp.c:229)
[2019-03-28 22:26:26.682] Port Forwarding (UPnP) Port forwarding through "http://192.168.1.1:5000/", service "". (local address: 192.168.1.247:57918) (/home/builder/.termux-bu>
[2019-03-28 22:26:26.682] Port Forwarding (UPnP) Port forwarding successful! (/home/builder/.termux-build/transmission/src/libtransmission/upnp.c:307)
[2019-03-28 22:26:26.682] Port Forwarding Forwarded (/home/builder/.termux-build/transmission/src/libtransmission/port-forwarding.c:92)
```

_OpenWrt log_
```
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: HTTP REQUEST from [::ffff:192.168.1.247]:60674 : GET /rootDesc.xml (HTTP/1.1)
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Host: 192.168.1.1:5000
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: HTTP REQUEST from [::ffff:192.168.1.247]:60676 : POST / (HTTP/1.1)
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Host: 192.168.1.1:5000
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: SOAPAction: #GetStatusInfo
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: HTTP REQUEST from [::ffff:192.168.1.247]:60678 : POST / (HTTP/1.1)
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Host: 192.168.1.1:5000
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: SOAPAction: #GetExternalIPAddress
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: HTTP REQUEST from [::ffff:192.168.1.247]:60680 : POST / (HTTP/1.1)
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Host: 192.168.1.1:5000
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: SOAPAction: #AddPortMapping
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: AddPortMapping: ext port 57918 to 192.168.1.247:57918 protocol TCP for: Transmission at 57918 leaseduration=604800 rhost=
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: UPnP permission rule 0 matched : port mapping accepted
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Check protocol tcp for port 57918 on ext_if eth1.2 76.175.66.15, 0F42AF4C
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: redirecting port 57918 to 192.168.1.247:57918 protocol TCP for: Transmission at 57918
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: HTTP REQUEST from [::ffff:192.168.1.247]:60682 : POST / (HTTP/1.1)
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Host: 192.168.1.1:5000
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: SOAPAction: #AddPortMapping
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: AddPortMapping: ext port 57918 to 192.168.1.247:57918 protocol UDP for: Transmission at 57918 leaseduration=604800 rhost=
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: UPnP permission rule 0 matched : port mapping accepted
Thu Mar 28 22:26:29 2019 daemon.debug miniupnpd[17092]: Check protocol udp for port 57918 on ext_if eth1.2 76.175.66.15, 0F42AF4C
Thu Mar 28 22:26:29 2019 daemon.info miniupnpd[17092]: redirecting port 57918 to 192.168.1.247:57918 protocol UDP for: Transmission at 57918
```

However it DOES NOT work when "enable additional logging" IS NOT checked:
```
[2019-03-28 22:38:12.688] Port Forwarding (UPnP) Found Internet Gateway Device "http://192.168.1.1:5000/" (/home/builder/.termux-build/transmission/src/libtransmission/upnp.c:>
[2019-03-28 22:38:12.688] Port Forwarding (UPnP) Local Address is "" (/home/builder/.termux-build/transmission/src/libtransmission/upnp.c:229)
[2019-03-28 22:38:12.688] Port Forwarding (UPnP) Port forwarding through "http://192.168.1.1:5000/", service "". (local address: :49482) (/home/builder/.termux-build/transmiss>
[2019-03-28 22:38:12.688] Port Forwarding Not forwarded (/home/builder/.termux-build/transmission/src/libtransmission/port-forwarding.c:92)
```

_OpenWrt log_
```
Thu Mar 28 22:38:15 2019 daemon.err miniupnpd[19417]: inet_aton([::ffff:192.168.1.247]:60710) FAILED
Thu Mar 28 22:38:15 2019 daemon.err miniupnpd[19417]: inet_aton([::ffff:192.168.1.247]:60712) FAILED
```

rodibin-at-placemaildotonline, 2019-04-13:
Updated maintainer @ldir-EDB0

neheb, 2019-04-13:
miniupnpd is unmaintained.

You can try running https://github.com/openwrt/packages/pull/8671

Alternatively, install strace and provide a log so that the error can be seen. Reporting to upstream is a better idea though.

rodibin-at-placemaildotonline, 2019-04-14:
Thanks for continuing to push patches to it even if it's unmaintained, you rock my world neheb!!!

Once 8671 is merged I will try that and report back. If the error still happens how do i report to upstream? Where is that

neheb, 2019-04-14:
https://github.com/miniupnp/miniupnp

rodibin-at-placemaildotonline, 2019-04-27:
Just to make sure I'm doing this right, how do I get this version to show up in opkg? 

adding this package feed doesn't seem to have the latest upnp?
http://downloads.openwrt.org/releases/packages-18.06/arm_cortex-a9_vfpv3/

neheb, 2019-04-27:
No, it is only available in snapshot. I think you can download the snapshok ipkg and install it using opkg.

rodibin-at-placemaildotonline, 2019-05-11:
Sorry I'm just getting to this now. Unfortunately I can't install this via the method you recommended. 

I have a pretty stable OpenWrt right now with all my packages configured and limited bugs, so I'd rather not reflash OpenWrt snapshot versions. Any other work around for this so I can test this package out?

```
root@OpenWrt:/tmp# opkg install miniupnpd_2.1.20190408-2_arm_cortex-a9_vfpv3.ipk
Upgrading miniupnpd on root from 2.1-1 to 2.1.20190408-2...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for miniupnpd:
 * 	libip4tc0
 * 	libip6tc0
 * 	libuuid1
 * opkg_install_cmd: Cannot install package miniupnpd.
```

neheb, 2019-05-11:
Install the other libraries first.

rodibin-at-placemaildotonline, 2019-05-11:
K, I got all packages except libxtables12, I can't find it here!
http://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/base/

Also the libuuid1 isn't working:

Installing libuuid1 (2.33-2) to root...
Collected errors:
 * check_data_file_clashes: Package libuuid1 wants to install file /usr/lib/libuuid.so.1
	But that file is already provided by package  * libuuid
 * check_data_file_clashes: Package libuuid1 wants to install file /usr/lib/libuuid.so.1.3.0
	But that file is already provided by package  * libuuid
 * opkg_install_cmd: Cannot install package libuuid1.

neheb, 2019-05-11:
No idea. Try to install with --force-depends

neheb, 2019-05-11:
Oh. it's a kmod i think.

rodibin-at-placemaildotonline, 2019-05-11:
Yeah I'm getting all sorts of overwrite errors with the other packages as well. Any way #8671 can be back ported to 18.06 and built or is that more effort than figuring this out?

neheb, 2019-07-02:
OpenWrt 19.07 will be out soon with that version.

thekiefs, 2019-07-08:
This issue also impacts me. I look forward to testing on 19.07.

neheb, 2020-01-22:
OpenWrt 19.07 is out.

thekiefs, 2020-01-22:
I just tried this on Android with Termux's transmission package. The port is forwarded correctly on 19.07. You can probably close this.

closed 2020-01-22 by neheb