openwrt/packages #8534: [meta] PKG_CPE_ID and security issue tracking

ja-pa, 2019-03-29:
Hi,
is there any documentation for maintainers how to handle PKG_CPE_ID ? Or some idea? 

@Andy2244 Pointed out in this PR  https://github.com/openwrt/packages/pull/8525#issuecomment-477935436  that it would be a useful thing to have some sort of documentation with examples  (and I agree with him).

Right now there are project already using this variable https://github.com/kkreitmair/cve-indicator

neheb, 2019-03-29:
beats me. I just copy CPE entries from nist.gov.

jefferyto, 2019-03-31:
This is probably obvious to other contributors, but I thought I'd add what I've gathered:

`PKG_CPE_ID` is the CPE ID (version 2.2) of the upstream project. No idea if version 2.3 IDs are accepted. Like @neheb, I usually search [nvd.nist.gov](https://nvd.nist.gov/vuln/search) or just Google to get the ID.

Pretty sure uscan (https://sdwalker.github.io/uscan/) uses these IDs to scan for new vulnerability reports. Package maintainers get an email when uscan finds CVEs that apply to their package. (uscan also sends emails when upstream has updated, prompting maintainers to update their packages. No idea what software uscan uses or how it works behind-the-scenes, other than it shares a name with a [Debian devscript](https://manpages.debian.org/unstable/devscripts/uscan.1.en.html).)

neheb, 2019-03-31:
uscan can sometimes figure out the CPE_ID but not always. Having it explicit helps ensure that it gets it right.

Andy2244, 2019-04-01:
So anyone cares to write a example with comments and explaining the mechanism?

I still don't know what exactly i'm supposed to-do.

BKPepe, 2019-05-17:
Let me a little re-use this issue, but would it be possible if there is fixed CVE in new version to include it in commit message?

BKPepe, 2019-08-09:
PKG_CPE_ID was added by @lynxis  maybe he can tell us something more about it? It was acked by @jow- . 

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c61a2395140d92cdd37d3d6ee43a765427e8e318

lynxis, 2019-08-09:
The idea behind the PKG_CPE_ID is to have a simple way to find the related CVE.

@sdwalker created uscan before PKG_CPE_ID. Somehow uscan figured out (or was the mapping created by hand?) which package relates to which CPE_ID / CVE. We asked ourself how can we help such CVE scanners. PKG_CPE_ID was the answer.

sdwalker, 2019-08-18:
PKG_CPE_ID gets piped into [cvechecker](https://github.com/sjvermeu/cvechecker) with the package version appended. cvechecker doesn't support 2.3 IDs so PKG_CPE_ID is mostly 2.2. [mailman](https://github.com/openwrt/packages/blob/master/mail/mailman/Makefile#L21) requires a 2.3->2.2 sed replacement.

The initial mappings were created by hand. https://gist.github.com/sdwalker/c4c674f2bd4f8321d8b427c7b50d24b0 is the remaining mappings. PKG_CPE_ID takes preference if it's added.

neheb, 2020-01-17:
Is this issue still relevant?

PKG_CPE_ID was added en masse with https://github.com/openwrt/packages/pull/10062

jefferyto, 2020-01-17:
I think this issue is about documentation? Perhaps someone would be kind enough to add a description of `PKG_CPE_ID` to the wiki? (https://openwrt.org/docs/guide-developer/packages and/or https://openwrt.org/docs/guide-developer/package-policies look like appropriate places)