openwrt/packages #8535: python-crypto: config symbol dependent = "Kconfig recursive dependency limitations"

Andy2244, 2019-03-29:
Maintainer: @jefferyto 
Environment: snapshots

Description: I can't find a way to define a `python-crypto` dependency, with a config symbol without running into:
`Kconfig recursive dependency limitations`. 

```
define Package/samba4-server
  $(call Package/samba4/Default)
  TITLE+= server
  DEPENDS:= +samba4-libs +SAMBA4_SERVER_AD_DC:python-crypto
endef
```
```
tmp/.config-package.in:25612:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:25612:   symbol PACKAGE_python-crypto is selected by PACKAGE_samba4-server
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:97437:   symbol PACKAGE_samba4-server depends on PACKAGE_python-crypto
```

In contrast something like this works fine:
`DEPENDS:= +samba4-libs +SAMBA4_SERVER_AD_DC:python-ldap`
or
`DEPENDS:= +samba4-libs +SAMBA4_SERVER_AD_DC:python-light`

I'm not sure how to resolve this problem and can ensure that for the AD-DC build option, the python-crypto are build and correctly installed alongside.

Any ideas?

Andy2244, 2019-03-29:
Here is a stripped down makefile to test this:
```
include $(TOPDIR)/rules.mk

PKG_NAME:=samba
PKG_VERSION:=4.9.5
PKG_RELEASE:=1

PKG_MAINTAINER:=Andy Walsh <andy.walsh44+github@gmail.com>
PKG_LICENSE:=GPL-3.0-only
PKG_LICENSE_FILES:=COPYING

PKG_SOURCE_URL:=https://download.samba.org/pub/samba/stable/
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_HASH:=078956d2d98e22011265afd4b7221efe4861067dcba4a031583b01f34d423700

include $(INCLUDE_DIR)/package.mk
include $(TOPDIR)/feeds/packages/lang/python/python-package.mk

PKG_CONFIG_DEPENDS:= \
	CONFIG_SAMBA4_SERVER_AD_DC

define Package/samba4/Default
  SECTION:=net
  CATEGORY:=Network
  TITLE:=Samba $(PKG_VERSION)
  URL:=http://www.samba.org/
endef

define Package/samba4-libs
  $(call Package/samba4/Default)
  TITLE+= libs
  DEPENDS:= +zlib +libtirpc +libpopt +libcomerr
endef

define Package/samba4-server
  $(call Package/samba4/Default)
  TITLE+= server
  DEPENDS:=	+samba4-libs +SAMBA4_SERVER_AD_DC:python-light +SAMBA4_SERVER_AD_DC:python-crypto
  # DEPENDS:=	+samba4-libs +python-crypto
  # DEPENDS:=	+samba4-libs +python-light +python-crypto
endef

$(eval $(call BuildPackage,samba4-libs))
$(eval $(call BuildPackage,samba4-server))
```

The only version i found that works is the last option, all other will fail:
`DEPENDS:=	+samba4-libs +python-light +python-crypto`

jefferyto, 2019-03-29:
I see from [how `SAMBA4_SERVER_AD_DC` is defined](https://github.com/openwrt/packages/blob/master/net/samba4/Config.in#L12-L28) in Config.in:

```
config SAMBA4_SERVER_AD_DC
	bool "Active Directory Domain Controller support (requires krb5-server) (EXPERIMENTAL)"
	depends on PACKAGE_samba4-server
	select PACKAGE_python-base
	select PACKAGE_python-crypto
	select PACKAGE_libopenssl
	select PACKAGE_libgnutls
	select PACKAGE_libopenldap
	help
		installs: samba (meta-daemon) python-crypt ntlm_auth
			scripts: samba-tool

		Run as a Active Directory Domain Controller
		see: https://wiki.samba.org/index.php/Setting_up_Samba_as_an_Active_Directory_Domain_Controller
		HINT: see section (# Using the Domain Controller as a File Server)
		NOTE: Extroot is recommend for this setup, as it is not optimized to run completely from RAM/tempfs!
	default n
```
It already selects python-crypto if `SAMBA4_SERVER_AD_DC` is selected.

If you just change `python-base` to `python-light`, would that do what you want?

Andy2244, 2019-03-29:
Yeah i cleaned up my Config.in file and atm i use this, which works:

`PKG_BUILD_DEPENDS:=samba4/host e2fsprogs/host SAMBA4_WAF_CROSS_EXECUTE:qemu-userspace/host`

```
config SAMBA4_SERVER_AD_DC
	bool "Active Directory Domain Controller support (requires krb5-server) (EXPERIMENTAL)"
	depends on PACKAGE_samba4-server
	select PACKAGE_python-crypto
	help
		installs: samba (meta-daemon) python-crypto ntlm_auth
			scripts: samba-tool

		Run as a Active Directory Domain Controller
		see: https://wiki.samba.org/index.php/Setting_up_Samba_as_an_Active_Directory_Domain_Controller
		HINT: see section (# Using the Domain Controller as a File Server)
		NOTE: Extroot is recommend for this setup, as it is not optimized to run completely from RAM/tempfs!
	default n
```

So just selecting `python-crypto` + all deps via menuconfig, the problem with only selecting via menuconfig is that any manual created configs or manual pkg-installs of samba4-server will not automatically install python-lite + python-crypto. Thats why i really want a true `DEPENDS:=	+samba4-libs +python-light +python-crypto`, so this can't fail.

jefferyto, 2019-03-29:
What happens if you remove `select PACKAGE_python-crypto` from Config.in?

BTW you didn't include the updated Config.in in #8353.

Andy2244, 2019-03-29:
> What happens if you remove `select PACKAGE_python-crypto` from Config.in?

Check my simple example above, there it does not use any Config.in at all and i run into the same problems as long as the depends is conditional.

jefferyto, 2019-03-29:
Because your example doesn't include Config.in, `SAMBA4_SERVER_AD_DC` is not defined. I don't know what make decides to do when there are undefined symbols.

Andy2244, 2019-03-29:
> Because your example doesn't include Config.in, `SAMBA4_SERVER_AD_DC` is not defined.

My bad, did forget to copy it in, was already using a minimal Config.in locally. Still getting the same errors.

cotequeiroz, 2019-03-29:
I have a hackish suggestion that does what you intend.
Since `Config.in` already selects packages for building, you may use `EXTRA_DEPENDS` to add them at install time, avoiding the recursive dependency mess.
```
define Package/samba4-server
  $(call Package/samba4/Default)
  TITLE+= server
  DEPENDS:= +samba4-libs
  EXTRA_DEPENDS:=$(if $(CONFIG_SAMBA4_SERVER_AD_DC),python-crypto,)
endef
```
Without `CONFIG_SAMBA4_SERVER_AD_DC`:
```
$ cat build_dir/target-mipsel_74kc_musl/samba-4.9.4/ipkg-mipsel_74kc/samba4-server/CONTROL/control
Package: samba4-server
Version: 4.9.4-3
Depends: libc, samba4-libs
```
After setting `CONFIG_SAMBA4_SERVER_AD_DC`:
```
$ cat build_dir/target-mipsel_74kc_musl/samba-4.9.4/ipkg-mipsel_74kc/samba4-server/CONTROL/control
Package: samba4-server
Version: 4.9.4-3
Depends: python-crypto, libc, samba4-libs
```
I tested by installing it, and it pulled up python-crypto OK:
```
Configuring libpam.
Configuring libdbus.
Configuring dbus.
dbus[11735]: Unknown group "netdev" in message bus configuration file
Configuring libavahi-dbus-support.
Configuring libdaemon.
Configuring avahi-dbus-daemon.
Configuring python-base.
Configuring libffi.
Configuring libbz2.
Configuring python-light.
Configuring python-codecs.
Configuring python-db.
Configuring libgpg-error.
Configuring libgcrypt.
Configuring libncurses.
Configuring libcomerr.
Configuring libss.
Configuring krb5-libs.
Configuring libavahi-client.
Configuring python-decimal.
Configuring libgmp.
Configuring libnettle.
Configuring python-distutils.
Configuring libattr.
Configuring attr.
Configuring python-pydoc.
Configuring python-ctypes.
Configuring python-multiprocessing.
Configuring python-logging.
Configuring python-sqlite3.
Configuring libgdbm.
Configuring python-gdbm.
Configuring python-email.
Configuring python-openssl.
Configuring python-xml.
Configuring python-compiler.
Configuring python-unittest.
Configuring python-ncurses.
Configuring python.
Configuring python-crypto.
Configuring libpopt.
Configuring samba4-libs.
Configuring samba4-server.
```

Andy2244, 2019-03-29:
Ah cool, will try this via EXTRA_DEPENDS thanks.

jefferyto, 2019-03-29:
I think this has something to do with how python-crypto's `DEPENDS` is set up:

```
DEPENDS:=+libgmp
...
DEPENDS+=+PACKAGE_python-crypto:python
```

Compared to python-ldap (which does not have this issue):

```
DEPENDS:=+libopenldap +python
```

@commodo have you seen similar issues before?

cotequeiroz, 2019-03-29:
I forgot to mention, but ensure that you're selecting the packages' dependencies, as select will not check them for you.  Use `tmp/.config-package.in` from when you got the recursive dependencies error to see what was added, then manually do it in `Config.in`, omitting the bad ones.

cotequeiroz, 2019-03-29:
> I think this has something to do with how python-crypto's DEPENDS is set up:

I quickly (haven't tried all combinations) checked that and it fails.

commodo, 2019-04-01:
This looks weird a bit.
I wouldn't select python-crypto in 2 places; i.e. here in  https://github.com/openwrt/packages/blob/master/net/samba4/Config.in#L16  &  in the `DEPENDS:= +samba4-libs +SAMBA4_SERVER_AD_DC:python-crypto`.
Doing that, does look a bit circular.

If adding a Config.in that selects `python-crypto` via `SAMBA4_SERVER_AD_DC` I think that is sufficient.
One thing you could try is to select `+@SAMBA4_SERVER_AD_DC`  in `samba4-server` but it still feels circular, because this is a boolean Kconfig flag that looks like it should be enabled by a user in menuconfig.

Andy2244, 2019-04-01:
> If adding a Config.in that selects `python-crypto` via `SAMBA4_SERVER_AD_DC` I think that is sufficient.

The problem is select only works in menuconfig, if you manually add or merge the config changes the dependency gets lost. I still would like to have the python packages behave as any other normal package, regarding dependencies.

commodo, 2019-04-01:
Would just adding `+@SAMBA4_SERVER_AD_DC` work ?

Andy2244, 2019-04-01:
No, what i need is that python-crypto gets installed build, if the config option SAMBA4_SERVER_AD_DC is present.

commodo, 2019-04-01:
`python-crypto` should be selected when `+@SAMBA4_SERVER_AD_DC` is added to the DEPENDS;
and then `SAMBA4_SERVER_AD_DC` select `python-crypto`

or, that is my understanding by looking at Config.in (for `SAMBA4_SERVER_AD_DC`)

Andy2244, 2019-04-01:
I don't get this, i don't want add SAMBA4_SERVER_AD_DC to all server builds, thats just a user option. If i add +@SAMBA4_SERVER_AD_DC than the server will always be build with ad_dc + python support?
Do i misunderstand "+@" ?

jow-, 2019-04-01:
The error is not strictly related to `python-crypto`, but happens due to the way transitive dependencies are aggregated and propagated among packages by the build system.

What makes `python-crypto` special compared with, e.g. `python-ldap` is that `python-crypto` depends on `python-light`. Python light in turn has a conditional dependency on `PYTHON3_BLUETOOTH_SUPPORT` which resolves into another bunch of conditional deps.

commodo, 2019-04-01:
> I don't get this, i don't want to add SAMBA4_SERVER_AD_DC to all server builds, thats just a user option. 

hmm, ok;
but then doing `DEPENDS:= +SAMBA4_SERVER_AD_DC:python-crypto`  looks un-needed, since you still need to enable SAMBA4_SERVER_AD_DC manually;
or maybe i missed some discussion from the initial post till here

cotequeiroz, 2019-04-02:
I was absent for a couple of days.  @jow-, python-crypto depends on python, not python-light.  python depends on python-light.  The rest of your explanation is in line with what I observed.  One thing in which they differ is that python-crypto uses a conditional to match the python version to depend on.  However, as I mentioned earlier, removing that conditional does not mitigate the issue.

My suggestion is to do the buildroot's job manually.  Please anyone correct me if I'm wrong, since I'm no expert in the buildroot.  So `select`ing `PACKAGE_python-crypto` in `Config.in` will build python-crypto.  However, it will not be recorded in the pakcage's `control` file as a dependency of `samba4-server`, so it will not be automatically installed by opkg.  This can be corrected by using the `EXTRA_DEPENDS`.  I agree that the correct way to fix this is to find where the dependency propagation logic is failing, but I will need to study it thoroughly, which will take a long time.

cotequeiroz, 2019-04-07:
I think I have a solution for these apparently random python circular dependencies.
TLDR: my conclusion is that the `USB_SUPPORT` symbol has to be selected along with `PYTHON[3]_BLUETOOTH_SUPPORT` before selecting `bluez-libs` in `python[3]-light`:
`+(PYTHON[3]_BLUETOOTH_SUPPORT&&USB_SUPPORT):bluez-libs`.
Otherwise, bluez-libs gets selected without `USB_SUPPORT` selected.  I'm not sure how to explain this properly.  Here's a tree of the python - python-light dependencies that I came up with.
```
python          
  +python-light        
    +bluez-libs if PYTHON_BLUETOOTH_SUPPORT      
      +kmod-bluetooth    
        @USB_SUPPORT  
        +kmod-usb-core  
          @USB_SUPPORT
        +kmod-crypto-hash  
        +kmod-crypto-ecb  
        +kmod-crypto-crc16  
        +kmod-hid  
        +kmod-crypto-cmac if !LINUX_3_18  
        +kmod-regmap-core if !LINUX_3_18  
        +kmod-crypto-ecdh if !LINUX_3_18 && !LINUX_4_9  
        +kmod-rfkill if USE_RFKILL  
          +kmod-input-core
          @USE_RFKILL
```
As one can see, if we define `PYTHON_BLUETOOTH_SUPPORT`, we select two packages that depend on `USB_SUPPORT`, without either selecting it or depending on it.  In theory, if we were to make `PYTHON_BLUETOOTH_SUPPORT` dependent on `USB_SUPPORT`, it should be resolved, but it does not.  I haven't studied the dependency resolver to figure out why exactly, but it apparently comes up  with a list of dependencies without optimizations to remove redundant or impossible dependencies.
It does work if we change `python[3]-light` `bluez` `DEPENDS` to 
```
+(PYTHON3_BLUETOOTH_SUPPORT&&USB_SUPPORT):bluez-libs
```
So that `bluez-libs` is only picked when `USB_SUPPORT` is on.  Of course, to be consistent, we can surround the Configuration menu for `python3-light` with `if USB_SUPPORT` and `endif`, so it only appears when we do have `USB_SUPPORT`.

If we do that, then the proposed code here (`DEPENDS:= +samba4-libs +SAMBA4_SERVER_AD_DC:python-crypto`)  works, and we can revert all commits in #8407 (and I'm betting more that removed the compilation-time optimization of building only the necessary python versions of the modules).
@jow-, please tell me what do you think of this.

cotequeiroz, 2019-04-15:
It won't work exactly as I expected, but there is still a way.  The `+(PYTHON3_BLUETOOTH_SUPPORT&&USB_SUPPORT) syntax is not supported by `include/package-ipkg.mk`, even though `make menuconfig` does the right thing.  So we can use `EXTRA_DEPENDS` to add the `bluez-libs` dependency, and it should work the same way.  But first I'll see if I can come up with a proper way to add the `&&` syntax to the dependency generator.

commodo, 2019-04-16:
only as a note: this all seems familiar, when trying to dig through DEPENDS

cotequeiroz, 2019-04-16:
@commodo 
Since you mentioned digging through DEPENDS, I'll share my thoughts with you.  If you ever get tired of the philosopher here, let me know :smile: 

From what I could infer:

- `tmp/.packageinfo` is the main source to build the dependencies, even though they are not all in there.

- `tmp/.packagedeps` is where  `make` figures out what needs to be build in what order, built from the `.packageinfo` contents.

- `tmp/.config-package.in` is where `make menuconfig` figures out what packages get selected, shown or hidden.

- Each package has its own `control` file which tells `opkg` what packages need to be installed as dependencies. 

`scripts/package-metadata.pl` is responsible to generate `.packagedeps` and `.config-package.in` from `.packageinfo`.  Finally `include/package-ipkg.mk` is where the individual packages' `control` files are generated.

- The packages `DEPENDS` setting works for all three dependency targets.

- The `package/<name>/config` definition/`Config.in` file contents are added to `.config-package.in`, so it helps to define the `make menuconfig` behavior.

- The packages `PKG_BUILD_DEPENDS` sets dependencies for `.packagedeps`.

- The packages `EXTRA_DEPENDS` setting adds them only to the ipk's `control` file.

Each of the three systems have different support levels for the rules' syntax:

- `.config-package.in` seems to be the most complete, where you can mix and match parentheses, `&&`s, `||`s, and `!`s.

- `.packagedeps` does not support `&&` (which I'm adding), supports `!` only at the beginning of the rule, and does not work with `||` either (which I'm adding as well), or parentheses (which I'm making it just ignore them).

- Package `config` generation logic supports `!` only at the beginning of the rule, completely ignores parentheses, and does not support `&&` (which I'm adding as well).

Did I get this right so far?

commodo, 2019-04-17:
> Did I get this right so far?

Pretty much. It's been a while since I went through them to remember this detail-level.

cotequeiroz, 2019-04-17:
Thanks for the feedback.
One more thing I forgot to mention, since this is kind of documenting the process some.
Generation of `.packagedeps` and `.config-package.in` is done before loading `.config`, so you can't use configuration symbols to make your own rules inside the package's Makefile, such as:
```
ifdef CONFIG_SOME_SYMBOL
    DEPENDS += +libsomething
    PKG_BUILD_DEPENDS +=, python/host
endef
```
or `DEPENDS=$(if $(CONFIG_SOME_SYMBOL), +libsomething)`.  They will work with `EXTRA_DEPENDS` though, since the package `control` file is generated at build-time, when `.config` has already been read.

I will, btw, add some notes to the developer guide.

Andy2244, 2019-05-16:
@cotequeiroz quick question, is there any update on this? Do i still use the `EXTRA_DEPENDS` or is there some new, better way to handle this?

cotequeiroz, 2019-05-16:
No news.  I'll open a PR shortly to add the EXTRA_DEPENDS, as it is currently the only way to do it.  You may monitor the current status of the patch to add && syntax here: [https://patchwork.ozlabs.org/project/openwrt/list/?series=104503&archive=both&state=*](https://patchwork.ozlabs.org/project/openwrt/list/?series=104503&archive=both&state=*).

cotequeiroz, 2019-05-20:
This has finally been fixed with #9018, so I'm closing it.  @Andy2244 try your change again; I used it to test #9018, and it should be working now.

closed 2019-05-20 by cotequeiroz

Andy2244, 2019-05-20:
ok thanks

Andy2244, 2019-12-05:
@cotequeiroz Hey i stumbled into this again maybe you can have a look?
[samba4.zip](https://github.com/openwrt/packages/files/3927913/samba4.zip)

Mainly happens when i try to add` zlib-dev` with `python3`.

```
define Package/samba4-libs
  $(call Package/samba4/Default)
  TITLE+= libs
  DEPENDS:= +zlib-dev +libtirpc +libpopt +libreadline +libcap +libgnutls +libopenssl +attr +acl \
	+PACKAGE_libpthread:libpthread +PACKAGE_libnettle:libnettle +PACKAGE_libgcrypt:libgcrypt +PACKAGE_libpam:libpam +PACKAGE_dbus:dbus +PACKAGE_libavahi-client:libavahi-client \
	+SAMBA4_SERVER_AVAHI:libavahi-client \
	+SAMBA4_SERVER_AD_DC:python3-dev +SAMBA4_SERVER_AD_DC:python3-crypto +SAMBA4_SERVER_AD_DC:libopenldap +SAMBA4_SERVER_AD_DC:jansson +SAMBA4_SERVER_AD_DC:libarchive
endef
```
with
```
 make menuconfig
Collecting package info: done
tmp/.config-package.in:29717:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29717:   symbol PACKAGE_python3-base is selected by PACKAGE_python3-light
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:32200:   symbol PACKAGE_python3-light is selected by PACKAGE_python3
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:29291:   symbol PACKAGE_python3 is selected by PACKAGE_python3-crypto
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:30319:   symbol PACKAGE_python3-crypto is selected by PACKAGE_samba4-libs
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:122473:  symbol PACKAGE_samba4-libs is selected by PACKAGE_samba4-admin
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:122429:  symbol PACKAGE_samba4-admin depends on PACKAGE_zlib
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:71395:   symbol PACKAGE_zlib is selected by PACKAGE_python3-base
```

cotequeiroz, 2019-12-06:
I'm not able to test this at the moment, but since `zlib-dev` depends on `zlib`, you must select `zlib` as well.  So add `+zlib` to `samba4-libs` `DEPENDS`, and it should do the trick.  Let me know if it work.
Cheers

Andy2244, 2019-12-07:
> you must select `zlib` as well

Thanks, yes this did the trick!