openwrt/packages #8539: stubby: Could not parse IPv6 anycast address from stubby.yml

stelas, 2019-03-29:
# stubby: Could not parse IPv6 anycast address from stubby.yml

Maintainer: @jonathanunderwood
Environment: OpenWrt 18.06.2

Description:

Adding IPv6 anycast resolver address:
```
# uci add stubby resolver
# uci set stubby.@resolver[-1].address='2001:67c:28a4::'
# uci set stubby.@resolver[-1].tls_auth_name='anycast.censurfridns.dk
```
Stubby terminates on service restart while parsing added address:
```
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: Scanner error: mapping values are not allowed in this context at line 22, column 33
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: Could not parse config file # Autogenerated configuration from uci data
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: resolution_type: GETDNS_RESOLUTION_STUB
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: round_robin_upstreams: 1
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: appdata_dir: "/var/lib/stubby"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: trust_anchors_backoff_time: 2500
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: tls_authentication: GETDNS_AUTHENTICATION_REQUIRED
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: tls_query_padding_blocksize: 128
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: edns_client_subnet_private: 1
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: idle_timeout: 10000
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: listen_addresses:
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - 127.0.0.1@5453
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - 0::1@5453
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: dns_transport_list:
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - GETDNS_TRANSPORT_TLS
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: upstream_recursive_servers:
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - address_data: 2a02:2970:1002::18
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:     tls_auth_name: "dns2.digitalcourage.de"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - address_data: 46.182.19.48
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:     tls_auth_name: "dns2.digitalcourage.de"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - address_data: 91.239.100.100
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:     tls_auth_name: "anycast.censurfridns.dk"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:   - address_data: 2001:67c:28a4::
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]:     tls_auth_name: "anycast.censurfridns.dk"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: , "Generic error"
Fri Mar 29 15:57:50 2019 daemon.err stubby[3747]: Could not parse config file "/var/etc/stubby/stubby.yml": Generic error
```

stelas, 2019-03-29:
Seems to be an issue in [upstream project](https://github.com/getdnsapi/stubby/issues/18).

stelas, 2019-03-29:
Some clarification.

jonathanunderwood, 2019-03-31:
So, I'm not sure there's much to be done here, right? Presumably I can close this.

neheb, 2019-03-31:
From my reading of the linked issue, the workaround is to use ::0.

jonathanunderwood, 2019-03-31:
> From my reading of the linked issue, the workaround is to use ::0.

Right. I could add checks to the init file which replace ipv6 addresses ending in `::` with ending in `::0`. But it feels like the wrong thing to do, given this is being fixed upstream anyway.

I think this can be closed.

closed 2019-03-31 by neheb