openwrt/packages #8547: unbound: list domain_forward "." results in (invalid) forward-zone: name: ".."

maertsen, 2019-03-30:
Maintainer: @EricLuehrsen 
Found in: unbound_1.9.1-1_x86_64.ipk
Environment: x86_64, apu4c2, 18.06.1

Thanks for this awesome packaging. I have been a happy user for quite a while.

Earlier today, I run into what I think is a bug. I'm trying to add a default forwarder to my upstream dns recursors.

Using 
```
config unbound
        list domain_forward '.'
```
in `/etc/config/unbound` results in a generated `/var/lib/unbound/unbound.conf` containing:
```
forward-zone:
  name: ".."
```
and unbound refuses to start.

I noticed the (very nice) rewrite of the forwarder functionality in master (408047628eb7a3605302e1ee410ddfbee01d5c83), which gets rid of domain_forward in the unbound section favor of zone statements. I wouldn't mind having that merged in the 18.06 branch instead of fixing this bug, though I understand if you don't :-)

EricLuehrsen, 2019-03-31:
I think merging "master" onto "openwrt-18.06" would be too much content change on a release branch. Master deletes the forward-zone single option and replaces it with many zones management.

EricLuehrsen, 2019-04-05:
Lets close this and continue any discussion in #8595

closed 2019-04-05 by BKPepe