openwrt/packages #8548: Buildbot failures

neheb, 2019-03-31:
Just putting this here for clarity:

node (including subpackages) - various platforms - https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/node-hid/compile.txt

noddos - all ( @cotequeiroz has a patch)

qemu - various - https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/qemu/compile.txt (has a libudev dependency)

softethervpn ( 64-bit targets ) - https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/softethervpn/compile.txt (faulty -m64 addition)

perl-device-usb - cortex a5 - https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a5_neon-vfpv4/packages/perl-device-usb/compile.txt

perl-inline - cortex a5 - https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a5_neon-vfpv4/packages/perl-inline/host-compile.txt

iotivity - cortex a5 https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a5_vfpv4/packages/iotivity/compile.txt (I think @hauke wanted to remove this one)

boost - cortex a5 - https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a5_vfpv4/packages/boost/compile.txt

gammu - cortex a8 - https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a8_neon/packages/gammu/compile.txt (libbluetooth dependency)

fio - mpcore - https://downloads.openwrt.org/snapshots/faillogs/arm_mpcore/packages/fio/compile.txt

php7 - armeb - https://downloads.openwrt.org/snapshots/faillogs/armeb_xscale/packages/

ffmpeg - i386 - https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/ffmpeg/full/compile.txt (could be a false-positive)

luasec - various - https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/luasec/compile.txt (seems to be related to fPIC vs fpic).

znc - various - https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/znc/compile.txt (seems to be related to fPIC vs fpic)

libgpg-error - mips64 - https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/libgpg-error/compile.txt (ping @MikePetullo )

protobuf - mips64 - https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/protobuf/compile.txt (weird linking error)

aircrack-ng - ppc - https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/aircrack-ng/compile.txt (ping @jbenden i think you said this was fixed?)

Will remove as they get fixed.

mhei, 2019-03-31:
The reason for failing php7 is failing dependencies. Just for the record, I regularly check build status for php7 and other packages which I maintain. I use [this ugly hacked script](https://gist.github.com/mhei/5f39c022003be4f80261046e3a7820a2) to get a quick overview. I wish I had more time to improve it and host it somewhere so that each package maintainer could use it.

BTW: I'm not sure whether such a cumulative issue is the best way to get things done and/or track it. Individual issue allow assigning them to persons/maintainers and close them when the issue is fixed.
But I don't want/like to spend much time on discussing how we solve it, just let's do it :smile:

cotequeiroz, 2019-04-02:
For the record.  The reason why I have not opened a PR for noddos is that I'm not sure the package works as it should, regardless of the patch, or the version of ipset.  See #8349.  Its main site, noddos.io, was down, many of the links to it, like the stats, and stats upload are still not working.  I do see it gathers a list of devices, but does not do much more than that.  Perhaps it is just me not knowing enough.  @StevenHessing maintains the package here, and upstream, and responded to #8349.  I submitted the patches upstream for him, and will probably wait for him to answer there before posting a PR here.  If you think I should open one now, so the package compiles, let me know.  Just beware that if a package fails to compile, then it fails to run; the opposite is not necessarily true.

tru7, 2019-04-03:
apcupsd: I was not aware of that part in the upstream source. It wants to configure the shutdown binary which will be called when battery status is low. That part is questionable for crosscompiling of course.

Would that be feasable in the Makefile: instead of
$(call Build/Configure/Default, \
use:
export SHUTDOWN=/sbin/halt && $(call Build/Configure/Default, \
??
Works on my toolchain.

jow-, 2019-04-03:
You could also use `CONFIGURE_VARS += SHUTDOWN=/sbin/halt` instead of redefining `Build/Configure`.

tru7, 2019-04-03:
@jow thanks for supporting a novice. Just verified. I will prepare a pull request tonight.

nxhack, 2019-04-03:
> node (including subpackages) - various platforms - https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/node-hid/compile.txt

If there is no objection, is it okay to take over those packages?

neheb, 2019-04-04:
I have none. Go for it. John seems to be busy with other stuff.

neheb, 2019-04-22:
Looks like most of them have been solved. The rest seem to be special.

cotequeiroz, 2019-04-22:
I will soon open a PR to fix noddos--as soon as I get some time.  @StevenHessing hasn't responded yet to noddos/noddos#53, but we have at least one user of the package who tested my fix; right now it does not build, so we can't even measure its download statics.  I'm still not sure it works, but if more people install it, we can have a better idea.

neheb, 2019-04-24:
@ClaymorePT Seems buildbots are failing on the boost update:

https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/domoticz/compile.txt
https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/i2pd/compile.txt

ClaymorePT, 2019-04-25:
@neheb -> Status code 404 (Not Found) on those links. Can't see the logs :/

neheb, 2019-04-25:
Looks like it's intermittent then.

ClaymorePT, 2019-04-25:
@neheb That seems to be an issue with ip2 code, not boost itself.

By looking  closer, we can see what's going on. There's a space between 'get_io_service' and '()'.

ClaymorePT, 2019-04-25:
@neheb That seems to be an issue with ip2 code, not boost itself.

neheb, 2019-04-25:
According to https://www.boost.org/doc/libs/1_70_0/doc/html/boost_asio/reference/io_service.html it seems to be deprecated. Meaning the packages need to be fixed.

ClaymorePT, 2019-04-25:
@neheb Yes, I was just looking into that. It was deprecated since 1.66.0 and it was finally removed in 1.70.0

neheb, 2019-04-25:
Added patches to fix them. Upstream already fixed.

neheb, 2019-05-02:
https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/openvswitch/compile.txt ping @yousong

yousong, 2019-05-03:
> https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/openvswitch/compile.txt ping @yousong

Thanks.  Will look into it these days.

yousong, 2019-05-03:
Hmm, will meed to backport this line [1] to openvswitch source code.  

 [1] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/net/openvswitch/conntrack.c?h=linux-4.14.y&id=5d827bfe37a5cecd1de33405afde4708657643c5

neheb, 2019-05-10:
Prometheus: https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/prometheus/compile.txt ping @aparcar @jefferyto 

https://github.com/fsnotify/fsnotify/commit/11054bcfedcc693de29270308bda9b4230791742#diff-9dee340ab29d1c1db9e272db38706348 Might fix this. fsnotify has a new version.

neheb, 2019-05-11:
@ammubhave libfolly is failing on several targets. Examples:

https://downloads.openwrt.org/snapshots/faillogs/armeb_xscale/packages/libfolly/compile.txt
https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/libfolly/compile.txt

@brunompena Missing dependency for tac_plus: https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/tac_plus/compile.txt

Should probably be made optional.

wongsyrone, 2019-05-12:
For FFmpeg issues stated above or https://github.com/openwrt/packages/issues/8686, you may try https://github.com/wongsyrone/packages-1/commit/5cad5c8568c58dbafd57c89ac4e89ec8495ae05d without the version bump, I come up with it when fixing LTO issues(many errors stated `plugin needed to handle lto object`), it doesn't use gcc-ar, gcc-nm and gcc-ranlib thus needs to specify LTO plugin. I fix the issue by passing values in rules.mk directly.

diizzyy, 2019-05-12:
@wongsyrone 
Have you tried compiling this on mips and mips64?

wongsyrone, 2019-05-12:
Nope, I don't have such devices now.

diizzyy, 2019-05-12:
I would expect that you need to do more modifications and platform fixes than just more or less bumping it.

neheb, 2019-05-12:
I doubt that's the issue. I think NASM is just broken. The error is that it's creating bad files.

diizzyy, 2019-05-12:
@neheb 
What I'm trying to say is that MIPS and other platforms gets wonky using OpenWrt's toolchain. I didn't bother tracking it down what changed in 4.X but I did however look at other distros including FreeBSD when I ran into various build/linking issues.

A few things are already handled in the "official" package however this below caused quite a few issues even if it built worked fine for ARM.
https://github.com/diizzyy/packages/blob/ffmpeg4/multimedia/ffmpeg4/Makefile#L62
https://github.com/diizzyy/packages/blob/ffmpeg4/multimedia/ffmpeg4/Makefile#L68

These are later (re)defined here:
https://github.com/diizzyy/packages/blob/ffmpeg4/multimedia/ffmpeg4/Makefile#L203
https://github.com/diizzyy/packages/blob/ffmpeg4/multimedia/ffmpeg4/Makefile#L205

This is why I doubt just bumping it will work flawlessy but I might of course be wrong.

wongsyrone, 2019-05-12:
> I think NASM is just broken

neheb, 2019-05-15:
@mwarning https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/smcroute/compile.txt

mwarning, 2019-05-15:
@neheb how can I reproduce the error? I've build smcroute for "Freescale MPC85xx" (powerpc_8540) without problems using the current openwrt master.

yousong, 2019-05-15:
> @neheb how can I reproduce the error? I've build smcroute for "Freescale MPC85xx" (powerpc_8540) without problems using the current openwrt master.

Select libcap and build it first, then rebuild `smcroute` should trigger it.

`--without-libcap` should fix the issue, https://github.com/troglobit/smcroute/blob/master/configure.ac#L55-L57

mwarning, 2019-05-15:
@yousong thanks
@neheb https://github.com/openwrt/packages/pull/8991

BKPepe, 2019-05-17:
Telldus-core

Maintainer: @PeterFromSweden

Faillog: https://downloads.openwrt.org/snapshots/faillogs/mipsel_74kc/packages/telldus-core/compile.txt

Seafile-server

Maintainer: @cotequeiroz 

Faillog: https://downloads.openwrt.org/snapshots/faillogs/mipsel_74kc/packages/seafile-server/compile.txt

Domoticz

Maintainer: @stintel 

Faillog: https://downloads.openwrt.org/snapshots/faillogs/mipsel_74kc/packages/domoticz/compile.txt
(most likely https://github.com/openwrt/packages/pull/8934 will fix it)

cotequeiroz, 2019-05-17:
> Faillog: https://downloads.openwrt.org/snapshots/faillogs/mipsel_74kc/packages/seafile-server/compile.txt

This should be fixed by https://github.com/openwrt/packages/pull/8986

neheb, 2019-05-17:
Seafile is dated May 16. Should go away soon.

Telldus is fixed by https://github.com/openwrt/packages/pull/8963 and https://github.com/openwrt/packages/pull/8996

Domoticz is fixed by https://github.com/openwrt/packages/pull/8934 . Needs to be merged.

neheb, 2019-05-18:
@ammubhave fbthrift seems to be failing.

https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/fbthrift/host-compile.txt

I've been able to reproduce the failure locally on arc platform.

neheb, 2019-05-19:
@MikePetullo snort3 does not compile on big endian targets. Incomplete fix:

```
--- a/src/codecs/ip/cd_ipv4.cc
+++ b/src/codecs/ip/cd_ipv4.cc
@@ -358,8 +358,8 @@ void Ipv4Codec::IP4AddrTests(
     /* Loopback traffic  - don't use htonl for speed reasons -
      * s_addr is always in network order */
 #ifdef WORDS_BIGENDIAN
-    msb_src = (iph.ip_src >> 24);
-    msb_dst = (iph.ip_dst >> 24);
+    msb_src = (iph->ip_src >> 24);
+    msb_dst = (iph->ip_dst >> 24);
 #else
     msb_src = (uint8_t)(iph->ip_src & 0xff);
     msb_dst = (uint8_t)(iph->ip_dst & 0xff);
```

neheb, 2019-05-20:
@nxhack https://downloads.openwrt.org/snapshots/faillogs/powerpc_464fp/packages/node-serialport/compile.txt

nxhack, 2019-05-21:
I will address this problem now. The musl header file for powerpc seems to be confused. Anyway, I will split the node-serialport package for patching "@serialport/bindings" npm module.

neheb, 2019-05-24:
@James-TR dnsdist seems to need a libatomic dependency: https://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/dnsdist/compile.txt

neheb, 2019-05-24:
@AmarOk1412 No idea if your recent update broke it, but xtables-addons is broken on mips64: https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/xtables-addons/compile.txt

neheb, 2019-05-27:
@cotequeiroz Any idea on this one? https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/libevhtp/compile.txt

Seems to be failing on all targets

James-TR, 2019-05-27:
@neheb I just had a look and dnsdist does indeed require libatomic on mips* and powerpc. Is it ok to add it as a standard dependency or is there a way to do it on a per-target level? https://github.com/PowerDNS/pdns/blob/9bb64785797d4b4829be3debc477458a7f81e7bd/m4/pdns_check_os.m4#L38-L44 is the relevant stanza in the preprocessing that pulls that in.

cotequeiroz, 2019-05-27:
> 
> 
> @cotequeiroz Any idea on this one? https://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/libevhtp/compile.txt
> 
> Seems to be failing on all targets

I can't test the exact failure case here as it builds OK.  Circle-ci has no trouble either.  I have two theories: (1) default openwrt-system include path are not being searched by default by the bot SDK; notice the final C_FLAGS shown; (2) oniguruma include files are not being copied to the right place.

For (1), we can add `TARGET_CPP_FLAGS` to `TARGET_CFLAGS`, which will explicitly add the include dirs; for (2), I just now noticed I used {} instead of () in the `SED` command; it should not have caused trouble, as the files are installed in the default search path, `$(STAGING_DIR)/usr/include`, and the library was found nonetheless.  Furthermore, libevhtp does not seem to use `ONIGURUMA_INCLUDE_DIR` anyway.

I've opened #9081 as an attempt to fix this.

cotequeiroz, 2019-05-27:
> @neheb I just had a look and dnsdist does indeed require libatomic on mips* and powerpc. Is it ok to add it as a standard dependency or is there a way to do it on a per-target level? [...]

You may use `DEPENDS+=+(mips||mipsel||mips64||mips64el||powerpc):libatomic`; you should add libatomic to `PKG_BUILD_DEPS` as well, since `tmp/.packagedeps` is not handling this syntax properly, so libatomic will not be included in the build deps otherwise.

neheb, 2019-05-27:
ARC needs libatomic as well:

https://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/dnsdist/compile.txt

I would just unconditionally include it. There's no harm. i386 and x86-64 have enough space to spare 5KB.

neheb, 2019-05-30:
@yousong Since you seem to maintain libunwind: https://downloads.openwrt.org/snapshots/faillogs/powerpc_8540/packages/libfolly/compile.txt

I looked into this and found this error all over the CMake log output:

```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_8540_gcc-9.1.0_musl/lib/gcc/powerpc-openwrt-linux-musl/9.1.0/../../../../powerpc-openwrt-linux-musl/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-powerpc_8540_musl/usr/lib/libunwind.so: undefined reference to `getcontext'
collect2: error: ld returned 1 exit status
```

I tried to work around it by including ucontext.h before everything but that did not work. This leads me to believe libunwind is the reason why it's failing on a bunch of targets.

yousong, 2019-05-30:
@neheb 

libunwind for powerpc depends on `getcontext()` from libc which musl-libc does not provide because these APIs are supposed to be obsolescent.  I will post a patch to libunwind indicate such dependency.  Thanks for the report.

yousong, 2019-05-30:
@neheb , libunwind patch applied, http://git.openwrt.org/1e5f4dcd66807b662f0ffb2bebd0c8bf9bf89c92

neheb, 2019-05-31:
@yousong different error on i386:

```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium_gcc-9.1.0_musl/bin/i486-openwrt-linux-musl-g++  -Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-i386_p$/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium_gcc-9.1.0_musl/lib/gcc/i486-openwrt-linux-musl/9.1.0/../../../../i486-openwrt-linux-musl/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-i386_pentium_musl/usr/lib/libunwind.so: undefined reference to `setcontext'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium_gcc-9.1.0_musl/lib/gcc/i486-openwrt-linux-musl/9.1.0/../../../../i486-openwrt-linux-musl/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-i386_pentium_musl/usr/lib/libunwind.so: undefined reference to `sigreturn'
collect2: error: ld returned 1 exit statu
```

neheb, 2019-06-01:
@cotequeiroz Just noticed libevhtp. The InstallDev section is Package/InstallDev instead of Build/InstallDev.

yousong, 2019-06-03:
> @yousong different error on i386:
>
> ```
> /home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium_gcc-9.1.0_musl/bin/i486-openwrt-linux-musl-g++  -Os -pipe -march=pentium-mmx -fno-caller-saves [...]

Thanks for the report.  Will look into the dependency on setcontext.  Thank you.

yousong, 2019-06-04:
Hi @neheb , do you still remember which package emits the above faillogs?

neheb, 2019-06-04:
libfolly

stintel, 2019-06-04:
Can we not use a single ticket for multiple issues please? This will grow to an unreadable mess and is sending out way too many notifications. Tickets are free, so I don't see a reason to not do 1 ticket per issue.

neheb, 2019-06-04:
Hrm OK. I will close. The initial idea was to fix the remaining ones.

closed 2019-06-04 by neheb