openwrt/packages #8552: Adblock : how to install 3.6.5

bast69, 2019-03-31:
Maintainer: @dibdot
Environment: 
Raspberry Pi Model B Rev 2
ARMv6-compatible processor rev 7 (v6l)
OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d) 

Description:
the last version available is 3.5.5-3 in the repo.
but the last commit is 3.6.5

how to install it ?

EricLuehrsen, 2019-03-31:
Generalization for all Packages: Check the difference between branches "master" and "openwrt-18.06." It is good practice to put updates on "master" only and wait for some bug reports (or absence) before sliding those updates onto release branches. Some content is too much change to add to an existing release. Only hot fixes like crashes and security issues get on release branches quickly.

dibdot, 2019-04-01:
@bast69 
Please check the snapshot download links in this forum post: https://forum.openwrt.org/t/adblock-support-thread/507

Download both files to your /tmp directory and install them with `opkg install <IPK File> --force-maintainer`. Finally remove your LuCI caches (`rm -rf /tmp/luci-*`).
For the new DNS Report feature you need package 'tcpdump-mini' as well (take it from usual 18.06 repo). For further questions please join the forum.

closed 2019-04-01 by dibdot