openwrt/packages #8594: pcre-8.42 configure: error: Invalid C++ compiler or C++ compiler flags

porroso, 2019-04-05:
I get "configure: error: Invalid C++ compiler or C++ compiler flags" when compiling on OSX. Last error free revision for me is r9750. How to resolve?

closed 2019-04-13 by porroso

linbox1968, 2019-04-14:
I am running into the same issue today with latest trunk

linbox1968, 2019-04-15:
How did you get it to work? I see you closed your comment from April 5

linbox1968, 2019-04-15:
@porroso How did you solve it to close the issue you had April 5

neheb, 2019-04-15:
Could be related to uclibc++. Try switching the c++ library to libstdcpp.

linbox1968, 2019-04-16:
Thanks  --> I am not familiar how to do this. Do I change the make file of pcre? I usual don't have to dig that deep into the source code.

linbox1968, 2019-04-16:
would I change the makefile?

This line

26 include $(INCLUDE_DIR)/uclibc++.mk 

what would I change it too?

neheb, 2019-04-16:
Remove it and change $(CXX_DEPENDS) to +libstdcpp

linbox1968, 2019-04-17:
Thanks that worked, but will I have to do this now all the time when I want to upgrade?

neheb, 2019-04-17:
The fact that that worked indicates something broken with your setup.

porroso, 2019-06-21:
> @porroso How did you solve it to close the issue you had April 5

@linbox1968 I'm sorry for very late reply, I simple switched to Debian since in Mojave I start to have problems compiling