openwrt/packages #8598: emailrelay: does not listen on IPv6

hmoffatt, 2019-04-05:
Maintainer: @fededim 
Environment: OpenWRT 18.06

Description:
I have emailrelay configured in proxy mode listening for remote clients, but it is listening on IPv4 only.

```
config emailrelay 'proxy'
	option enabled '1'
	option mode 'proxy'
	option smarthost 'somehost'
	option port '25'
	option remote_clients '1'
```

fededim, 2019-04-05:
Very strange, [EmailRelayReference](http://emailrelay.sourceforge.net/#reference_md_IP_addresses) clearly states:

"By default the E-MailRelay server listens for connections on the wildcard IPv4 and IPv6 addresses, and when making outgoing connections it does not explicitly bind any address to the the local socket."

If you try to launch from shell adding the --interface :: option (it will force binding only to ipv6) does it work ?

 E.g. emailrelay --as-proxy=somehost:port --port 25  --interface ::

It would be useful to have also the log (without interface option), 

e.g. emailrelay --as-proxy=somehost:port --port 25  --log --verbose

hmoffatt, 2019-04-05:
I see it running as
```/usr/bin/emailrelay --no-daemon --as-proxy somehost --port 25 --remote-clients --client-tls --client-auth /etc/emailrelay.auth```

but netstat shows it's only listening on IPv4:

```
# netstat -anp | grep :25
tcp        0      0 0.0.0.0:25              0.0.0.0:*               LISTEN      2658/emailrelay
```

When I add "--interface ::" it segfaults! That option is not listed in the help either.

```
# /usr/bin/emailrelay --no-daemon --as-proxy somehost --port 25 --remote-clients --client-tls --client-auth /etc/emailrelay.auth --interface ::
Segmentation fault
```

All the log says is

```Fri Apr  5 19:23:08 2019 mail.info [30204]: emailrelay: info: smtp server on 0.0.0.0:25```

neheb, 2020-01-22:
I cannot reproduce:

```
root@OpenWrt:/# netstat -anp | grep :25
tcp        0      0 0.0.0.0:25              0.0.0.0:*               LISTEN      2867/emailrelay
tcp        0      0 :::25                   :::*                    LISTEN      2867/emailrelay
```

My feeling is somehost is somehow faulty. Closing.

closed 2020-01-22 by neheb

hmoffatt, 2020-01-22:
It's not a problem with somehost, it's affecting inbound connections to the proxy.

Maybe it is fixed in 19.07, did the package get upgraded? Definitely not working in 18.06.5.

neheb, 2020-01-22:
I believe so.

hmoffatt, 2020-08-02:
Just to follow up, I finally upgraded to OpenWRT 19.07.3 and now mailrelay is listening on IPv6 also.