openwrt/packages #8648: Freeradius3-common: missing dependencies for the following libraries: libpcre.so.1

sharikov, 2019-04-10:
Maintainer: Lucile Quirion lucile.quirion@savoirfairelinux.com
Environment: OpenWrt HEAD

Description:

Freeradius3-common fails to build. 

> Package freeradius3-common is missing dependencies for the following libraries:
libpcre.so.1<

Tried again with fresh install of Ubuntu 18.04.2 LTS and new OpenWrt build, and it fails again.

neheb, 2019-04-11:
Ping @yangfl

yangfl, 2019-04-13:
No idea. `make package/freeradius3/compile` fine on openwrt-sdk-18.06.2-ar71xx-generic_gcc-7.3.0_musl.Linux-x86_64

Robby-, 2019-04-13:
@yangfl: I'm experiencing this same problem on the master branch, which is probably what the OP meant.
I'm compiling on Debian 9.8, however.

yangfl, 2019-04-13:
Debian sid, but I don't think this could be related to build os. Adding a dep will probably work, but I won't have time/env to test that.

neheb, 2019-04-13:
Could not reproduce on Ubuntu 18.04. Might be some oddity with freeradius where it's picking up some host library.

Robby-, 2019-04-13:
Well, I can test that because I was preparing a PR to update freeradius3 to the latest stable version anyway.

I'm a bit confused because there is already a dependency set on it via `+PACKAGE_libpcre:libpcre` and although it seems erroneous, it did compile fine with this about 8 months ago, not sure what made it break as I can't seem to find anything related. However, I'll include a fix for this issue in my PR.

neheb, 2019-04-14:
There's a similar issue on the buildbots where gammu has a similar dependency on bluez-libs.

My theory is that there's a race condition since the buildbots run their make jobs in parallel. Basically libpcre gets built after the dependencies get resolved and freeradius picks it up.

When it comes to this dependency, it should be either set to off or on.

Robby-, 2019-04-16:
@neheb: I submitted a PR (#8681) that includes a fix for this issue. If you could take a look at it please.

closed 2019-04-17 by neheb

sharikov, 2019-04-17:
Thanks so much everyone and thanks for the fix @Robby-.

PR #8681 (2019-04-17): freeradius3: update to 3.0.19, workaround fifo error, and fix libpcre dependency
Maintainer: no maintainer at this time
Compile tested: ARM, Linksys EA3500, master
Run tested: ARM, Linksys EA3500, master

Description:
- Fix the libpcre dependency in Makefile. This closes #8648.
- Workaround the fifo error introduced by the update to 3.0.18
- Update to 3.0.19: Latest stable release, contains security fixes for EAP-PWD.