openwrt/packages #8663: wget: 1.20.3 seems to depend on openssl-util for https://url

builker, 2019-04-12:
No problem for http://url
```
root@OpenWrt:~# wget http://www.google.com -O /tmp/wget.out
--2019-04-11 19:20:03--  http://www.google.com/
Resolving www.google.com... 216.58.195.228, 2607:f8b0:4005:808::2004
Connecting to www.google.com|216.58.195.228|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: '/tmp/wget.out'

/tmp/wget.out           [ <=>                ]  10.96K  --.-KB/s    in 0s

2019-04-11 19:20:03 (48.2 MB/s) - '/tmp/wget.out' saved [11221]
```
A problem for **https**://url
```
root@OpenWrt:~# wget http://www.google.com -O /tmp/wget.out
--2019-04-11 19:20:03--  http://www.google.com/
Resolving www.google.com... 216.58.195.228, 2607:f8b0:4005:808::2004
Connecting to www.google.com|216.58.195.228|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: '/tmp/wget.out'

/tmp/wget.out           [ <=>                ]  10.96K  --.-KB/s    in 0s

2019-04-11 19:20:03 (48.2 MB/s) - '/tmp/wget.out' saved [11221]

root@OpenWrt:~# wget -d https://www.google.com -O /tmp/wget.out
Setting --output-document (outputdocument) to /tmp/wget.out
Setting --output-document (outputdocument) to /tmp/wget.out
DEBUG output created by Wget 1.20.3 on linux-gnu.

Reading HSTS entries from /root/.wget-hsts
--2019-04-11 19:20:15--  https://www.google.com/
Disabling SSL due to encountered errors.
```
A solution:
```
root@OpenWrt:~# opkg install openssl-util
Installing openssl-util (1.1.1b-3) to root...
Downloading file:///root/openwrt/r9844-1f839c8/packages/arm_xscale/base/openssl-util_1.1.1b-3_arm_xscale.ipk
Installing libopenssl-conf (1.1.1b-3) to root...
Downloading file:///root/openwrt/r9844-1f839c8/packages/arm_xscale/base/libopenssl-conf_1.1.1b-3_arm_xscale.ipk
Configuring libopenssl-conf.
Configuring openssl-util.

root@OpenWrt:~# wget https://www.google.com -O /tmp/wget.out
--2019-04-11 19:21:29--  https://www.google.com/
Resolving www.google.com... 216.58.195.228, 2607:f8b0:4005:808::2004
Connecting to www.google.com|216.58.195.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: '/tmp/wget.out'

/tmp/wget.out           [ <=>                ]  11.02K  --.-KB/s    in 0.001s

2019-04-11 19:21:29 (7.84 MB/s) - '/tmp/wget.out' saved [11285]
```

builker, 2019-04-12:
Actually package libopenssl-conf alone is enough.

neheb, 2019-04-13:
Hrm ping @cotequeiroz 

ENGINE support will come soon. But it seems libopenssl-conf only gets selected when specific engines are selected.

cotequeiroz, 2019-04-22:
This was due to a bug in openssl 1.1.1b.  It has been fixed just now, so I'm closing this.

closed 2019-04-22 by cotequeiroz