openwrt/packages #8667: Error relocating /usr/bin/php-cli: explicit_bzero: symbol not found

tianye, 2019-04-12:
Error relocating /usr/bin/php-cli: explicit_bzero: symbol not found

tianye, 2019-04-12:
root@ZBT:/etc/php7# opkg files 'libc'
Package libc (1.1.19-1) is installed on root and has the following files:
/lib/ld-musl-mipsel-sf.so.1
/lib/libc.so
/usr/bin/ldd
root@ZBT:/etc/php7# opkg files 'uclibcxx'
Package uclibcxx (0.2.4-3) is installed on root and has the following files:
/usr/lib/libuClibc++.so.0
/usr/lib/libuClibc++-0.2.4.so
root@ZBT:/etc/php7#

mhei, 2019-04-13:
Which PHP packages do you have installed?
Did you have compiled on you own, or use precompiled packages (from which release or master)? Maybe you could attach your diffconfig so that I can try to rebuild and verify on my own...

tianye, 2019-04-15:
Only so much:
opkg install php7-cli

tianye, 2019-04-15:
BusyBox v1.27.2 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r6422-87606e2
 -----------------------------------------------------

tianye, 2019-04-15:
root@ZBT:~#  cat /proc/version 
Linux version 4.9.86 (root@ubuntu) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r6422-87606e2) ) #0 Sun Mar 11 22:42:38 2018
root@ZBT:~#

tianye, 2019-04-15:
Linux version 4.9.86 (root@ubuntu) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r6422-87606e2) ) #0 Sun Mar 11 22:42:38 2018

mhei, 2019-05-01:
I need the output of `opkg list_installed` and `cat /etc/os-release`, not the kernel version.

mhei, 2019-05-16:
Timeout.

closed 2019-05-16 by mhei