openwrt/packages #8721: samba4: build fails with parallel jobs

dbermond, 2019-04-18:
**Maintainer:** @Andy2244

**Environment:**
- **Architecture:** arm_cortex-a15_neon-vfpv4
- **Model:** R7800
- **OpenWrt and packages:** git master

samba4 builds only if I use `-j1` with make. It fails to build with multiple jobs for me, resulting in this message:

```
*** [Makefile:474: /storage/linux/abs/openwrt-r7800-git/src/openwrt/build_dir/hostpkg/samba-4.9.6/.built] Error 1
make[3]: Leaving directory '/storage/linux/abs/openwrt-r7800-git/src/openwrt-packages/net/samba4'
time: package/feeds/packages/samba4/host-compile#71.11#26.57#134.18
make[2]: *** [package/Makefile:107: package/feeds/packages/samba4/host/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
```

- This seems to be caused by commit https://github.com/openwrt/packages/commit/3e1d9c0bfdcdd49500fe14c58bdb4e831fe8204a.

- I can confirm that it was building fine with multiple jobs up until commit https://github.com/openwrt/packages/commit/77436747ceb06e6686045ac832324aa48c76bc43.

- When reverting commits https://github.com/openwrt/packages/commit/2cf952f7bb28acfce02943a5d8c95f60e9f652c0 and https://github.com/openwrt/packages/commit/3e1d9c0bfdcdd49500fe14c58bdb4e831fe8204a on top of git master, it builds fine with multiple jobs.

It's hard to tell which output is coming from samba4 when using multiple make jobs, since everything is building and outputting in parallel, but I think these two blocks of output comes from it:

```
/bin/sh: rpcgen: command not found
[163/218] Generating INFILE_smb.service
[164/218] Generating INFILE_nmb.service
[165/218] Generating INFILE_winbind.service
[166/218] Generating INFILE_samba.service
/bin/sh: rpcgen: command not found
```

```
Build failed:  -> task failed (err #127): 
        {task: nfs41acl-h nfs41acl.x -> nfs41acl.h}
```

dbermond, 2019-04-19:
A workaround is to install `rpcgen` on the host system. By doing this, it builds fine with multiple jobs.

Somehow, it seems that [nfs-kernel-server](https://github.com/openwrt/packages/tree/master/net/nfs-kernel-server) (the package that provides `rpcgen`) is not being built/installed at the correct time for me.

lucize, 2019-04-20:
maybe PKG_HOST_DEPENDS:=nfs-kernel-server/host so before samba4/host will be ready the rpcgen will be available

Andy2244, 2019-04-23:
> PKG_HOST_DEPENDS:=nfs-kernel-server/host so before samba4/host

Does this work as a fix?

lucize, 2019-04-24:
the strange thing is that rpcgen is a dependency for gcc in debian 

```
 dpkg -S rpcgen
libc-dev-bin: /usr/share/man/man1/rpcgen.1.gz
libc-dev-bin: /usr/bin/rpcgen
```
but I was wrong (misspelled the variable), the correct patch is this

```
diff --git a/net/samba4/Makefile b/net/samba4/Makefile
index 86f0fb6..58f4f8e 100644
--- a/net/samba4/Makefile
+++ b/net/samba4/Makefile
@@ -19,7 +19,8 @@ PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
 PKG_HASH:=c9205a651a83d69e200fec9dd65e9fa360f0c75ab3275b3dcb74e5cbaec60807

 # samba4=(asn1_compile) e2fsprogs=(compile_et) nfs-kernel-server=(rpcgen)
-PKG_BUILD_DEPENDS:=samba4/host e2fsprogs/host nfs-kernel-server/host
+HOST_BUILD_DEPENDS:=nfs-kernel-server/host
+PKG_BUILD_DEPENDS:=samba4/host e2fsprogs/host

 PKG_CONFIG_DEPENDS:= \
        CONFIG_SAMBA4_SERVER_NETBIOS \

```

dbermond, 2019-04-24:
> > PKG_HOST_DEPENDS:=nfs-kernel-server/host so before samba4/host

> Does this work as a fix?

@Andy2244  No, this does not work. The building with multiple jobs still fails with this.

> the strange thing is that rpcgen is a dependency for gcc in debian

@lucize  I'm building on Arch Linux. The package that provides `rpcgen` ([rpcsvc-proto](https://www.archlinux.org/packages/extra/x86_64/rpcsvc-proto/)) is not a dependency for [gcc](https://www.archlinux.org/packages/core/x86_64/gcc/) on Arch.

> but I was wrong (misspelled the variable), the correct patch is this
> 
 ```
diff --git a/net/samba4/Makefile b/net/samba4/Makefile
index 86f0fb6..58f4f8e 100644
--- a/net/samba4/Makefile
+++ b/net/samba4/Makefile
@@ -19,7 +19,8 @@ PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
 PKG_HASH:=c9205a651a83d69e200fec9dd65e9fa360f0c75ab3275b3dcb74e5cbaec60807

 # samba4=(asn1_compile) e2fsprogs=(compile_et) nfs-kernel-server=(rpcgen)
-PKG_BUILD_DEPENDS:=samba4/host e2fsprogs/host nfs-kernel-server/host
+HOST_BUILD_DEPENDS:=nfs-kernel-server/host
+PKG_BUILD_DEPENDS:=samba4/host e2fsprogs/host

 PKG_CONFIG_DEPENDS:= \
        CONFIG_SAMBA4_SERVER_NETBIOS \
 ```
@lucize This patch works for me. It fixes building with multiple jobs when `rpcgen` is not installed on the host system. Thank you for this.

Andy2244, 2019-04-24:
ok will pick the patch than thanks @lucize

closed 2019-04-24 by hnyman