openwrt/packages #8732: nsd still being maintained?

n8v8R, 2019-04-20:
@acinonyx  

Having looked at the package history (current and master) with version 4.1.13 and compared to upstream version 4.1.27 I am curious thus whether the package is still actively maintained?

neheb, 2019-04-20:
The answer is no

n8v8R, 2019-04-20:
There are other packages that appear not being maintained any more. Why would such packages are not being marked as outdated/unmaintained, respectively being carried in the repo at all?

neheb, 2019-04-20:
Would you like to take over? The package compiles.

n8v8R, 2019-04-20:
I would not mind if I would have the resource to compile/build.

But that was not really question but rather why those package are not being marked as outdated/unmaintained - which would save opening this kind of thread?
Respectively being carried in the repo at all considering that features/security patches being missed and notwithstanding potential compatibility issues?

acinonyx, 2019-04-21:
If you could submit an upgrade patch, I would be happy to compile and test it.

n8v8R, 2019-04-21:
is it not compiled from upstream srouce https://www.nlnetlabs.nl/projects/nsd/download/ ?

neheb, 2019-04-21:
He means a patch to the OpenWrt Makefile.

closed 2019-08-06 by BKPepe