openwrt/packages #8742: uwsgi a process holding a robust mutex died.

peterwillcn, 2019-04-21:
# Issue template (remove lines from top till here)

Maintainer: @dengqf6 
Environment: OpenWrt SNAPSHOT r9886-399aa0b / LuCI openwrt-18.06 branch (git-19.079.57770-b99e77d)

```
Sun Apr 21 19:41:23 2019 daemon.info uwsgi: [deadlock-detector] a process holding a robust mutex died. recovering...
Sun Apr 21 19:41:43 2019 daemon.info uwsgi: [deadlock-detector] a process holding a robust mutex died. recovering...
Sun Apr 21 19:42:39 2019 daemon.info uwsgi: Sun Apr 21 19:42:39 2019 - worker 2 (pid: 22649) is taking too much time to die...NO MERCY !!!
Sun Apr 21 19:42:40 2019 daemon.info uwsgi: DAMN ! worker 2 (pid: 22649) died, killed by signal 9 :( trying respawn ...
Sun Apr 21 19:42:40 2019 daemon.info uwsgi: Respawned uWSGI worker 2 (new pid: 18736)
Sun Apr 21 19:42:49 2019 daemon.info uwsgi: Sun Apr 21 19:42:49 2019 - worker 3 (pid: 22783) is taking too much time to die...NO MERCY !!!
Sun Apr 21 19:42:50 2019 daemon.info uwsgi: DAMN ! worker 3 (pid: 22783) died, killed by signal 9 :( trying respawn ...
Sun Apr 21 19:42:50 2019 daemon.info uwsgi: Respawned uWSGI worker 3 (new pid: 18749)
Sun Apr 21 19:43:49 2019 daemon.info uwsgi: Sun Apr 21 19:43:49 2019 - worker 1 (pid: 22754) is taking too much time to die...NO MERCY !!!
Sun Apr 21 19:43:50 2019 daemon.info uwsgi: DAMN ! worker 1 (pid: 22754) died, killed by signal 9 :( trying respawn ...
Sun Apr 21 19:43:50 2019 daemon.info uwsgi: Respawned uWSGI worker 1 (new pid: 18823)

```

neheb, 2020-01-17:
Is this still an issue?

neheb, 2020-01-24:
ping @peter-stadler any idea?

peter-stadler, 2020-01-24:
No, not really, maybe not enough resouces … Does this happen regularly and if yes, when starting uwsgi or at some specific interaction? Is this with LuCI or another app that is served by uWSGI?

In the meantime, we changed the configuration of uwsgi quite a bit, too, and the logging would be different now, can you post the current output (logread | grep uwsgi)?

**NB:** The `daemon.err` (not `daemon.info`) just means that the emperor is printing messages to stderr (not stdout), there are e.g. about 20 information lines for a `service uwsgi restart` …

peter-stadler, 2020-01-31:
@peterwillcn, can you check if you experience this issue again?

neheb, 2020-02-23:
Closing due to timeout. There were also many changes since the report. Please reopen if still an issue.

closed 2020-02-23 by neheb

yogeshyadav2912, 2020-04-27:
[deadlock-detector] a process holding a robust mutex died. recovering...
!!! uWSGI process 208957 got Segmentation Fault !!!
*** backtrace of 208957 ***
/opt/virtualenv/marketfinder/bin/uwsgi(uwsgi_backtrace+0x2e) [0x482b1e]
/opt/virtualenv/marketfinder/bin/uwsgi(uwsgi_segfault+0x21) [0x482eb1]
/lib64/libc.so.6(+0x36340) [0x7f24a9520340]

peter-stadler, 2020-04-27:
In my understanding this is a problem in the application or between it and `uwsgi`. It is difficult to say something without knowing the `<app_name>`. I found some [hints for Django](https://stackoverflow.com/questions/53498240/uwsgi-segmentation-fault-when-serving-a-django-application) but some should apply also for other programs …
As you are in `/opt/virtualenv/`: Does it use the same `uwsgi` binary as provided by opkg?

yogeshyadav2912, 2020-04-27:
the app_name is python app and it to specific to our company. we are using uwsgi 2.0.18 in virtualenv as pip package

yogeshyadav2912, 2020-04-27:
we are getting segmentation fault at time of exit of this python <app_name>

peter-stadler, 2020-04-27:
I see, sorry I do not know anything about the pip version of uWSGI (and if it using a preinstalled opkg binary of uwsgi). Does the [official test](https://uwsgi-docs.readthedocs.io/en/latest/tutorials/Django_and_nginx.html#basic-test) works?

Else, maybe it is worth a try using the opkg version (as it is compiled for the specific hardware) and adding [`virtualenv=…` parameter](https://uwsgi-docs.readthedocs.io/en/latest/WSGIquickstart.html#virtualenvs) parameter. But, note this is a [modular build](https://uwsgi-docs.readthedocs.io/en/latest/Install.html#modular-builds), you will have to use the needed plugins, too: `opkg install uwsgi uwsgi-python3-plugin …` and add the appropriate [`plugin=python,…` prameters](https://uwsgi-docs.readthedocs.io/en/latest/WSGIquickstart.html#installing-uwsgi-with-python-support).

Right now there are only three precompiled plugins: 
* uwsgi-python3-plugin
* uwsgi-syslog-plugin
* uwsgi-logfile-plugin
We could propose more (or all) plugins for the upstream version, but it would take some time until it is approved and becomes available in the tree …

peter-stadler, 2020-05-03:
I would take a look at this setup of uWSGI if you can tell me how you get it from a clean install: 
On which platform are you working?
From which source you get python and pip in which version?
How do you install virtualenv and uwsgi in it?
Which C compiler do you use (if you compile uwsgi)?
Do you have a minimal python program showing the error?