openwrt/packages #8748: ntpclient never restart working while procps-ng-ps is installed

zhanhb, 2019-04-22:
https://github.com/openwrt/packages/blob/ffdbc7a637df8e3f39dd6916405f78840edccfb4/net/ntpclient/files/ntpclient.hotplug#L69
Command `ps` will got once procps-ng-ps is installed.
```
  PID TTY          TIME CMD
13773 pts/0    00:00:00 bash
20022 pts/0    00:00:00 ps
```

zhanhb, 2019-04-22:
Maybe busybox ps should be used here.

neheb, 2019-04-22:
Could also replace it with ps aux.

zhanhb, 2019-04-22:
I execute `ps w`, got as following.
```
  PID TTY      STAT   TIME COMMAND
  476 ttyS0    Ss+    0:00 /sbin/askfirst /usr/libexec/login.sh
 9947 pts/1    Ss     0:00 -ash
23382 pts/1    R+     0:00 ps w
```

neheb, 2019-04-22:
Guess the BusyBox solution is the one.

neheb, 2019-04-25:
@zhanhb please test https://github.com/openwrt/packages/pull/8803

zhanhb, 2019-04-25:
I replace the script file, work as expected. No issue for package `ntpclient`.

closed 2019-07-02 by neheb