openwrt/packages #8756: samba4/libtirpc: fail to build on macos

ldir-EDB0, 2019-04-22:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: x86, master

Description:

Enabling samba4 reveals a build failure in libtirpc in macos environment:

make[3]: Entering directory `/Users/kevin/wrt/feeds/packages/libs/libtirpc'
CFLAGS="-O2 -I/Users/kevin/wrt/staging_dir/host/include -I/Users/kevin/wrt/staging_dir/hostpkg/include -I/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/include -DGQ" CPPFLAGS="-I/Users/kevin/wrt/staging_dir/host/include -I/Users/kevin/wrt/staging_dir/hostpkg/include -I/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/Users/kevin/wrt/staging_dir/host/lib -L/Users/kevin/wrt/staging_dir/hostpkg/lib -L/Users/kevin/wrt/staging_dir/target-x86_64_musl/host/lib" /Applications/Xcode.app/Contents/Developer/usr/bin/make -j1 -C /Users/kevin/wrt/build_dir/hostpkg/libtirpc-1.1.4  
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       libtirpc_la-auth_none.lo
auth_none.c:81:9: error: unknown type name 'mutex_t'
        extern mutex_t authnone_lock;
               ^
auth_none.c:83:2: warning: implicit declaration of function 'mutex_lock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_lock(&authnone_lock);
        ^
auth_none.c:87:4: warning: implicit declaration of function 'mutex_unlock' is invalid in C99 [-Wimplicit-function-declaration]
                        mutex_unlock(&authnone_lock);
                        ^
auth_none.c:103:2: warning: implicit declaration of function 'mutex_unlock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_unlock(&authnone_lock);
        ^
auth_none.c:113:9: error: unknown type name 'mutex_t'
        extern mutex_t authnone_lock;
               ^
auth_none.c:117:2: warning: implicit declaration of function 'mutex_lock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_lock(&authnone_lock);
        ^
auth_none.c:123:2: warning: implicit declaration of function 'mutex_unlock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_unlock(&authnone_lock);
        ^
auth_none.c:166:9: error: unknown type name 'mutex_t'
        extern mutex_t ops_lock;
               ^
auth_none.c:170:2: warning: implicit declaration of function 'mutex_lock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_lock(&ops_lock);
        ^
auth_none.c:180:2: warning: implicit declaration of function 'mutex_unlock' is invalid in C99 [-Wimplicit-function-declaration]
        mutex_unlock(&ops_lock);
        ^
7 warnings and 3 errors generated.
make[6]: *** [libtirpc_la-auth_none.lo] Error 1
make[5]: *** [all-recursive] Error 1
make[4]: *** [all] Error 2
make[3]: *** [/Users/kevin/wrt/build_dir/hostpkg/libtirpc-1.1.4/.built] Error 2
make[3]: Leaving directory `/Users/kevin/wrt/feeds/packages/libs/libtirpc'
time: package/feeds/packages/libtirpc/host-compile#0.24#0.18#0.48

diizzyy, 2019-04-22:
Having a quick look it seems that we'd need to make MacOS specific hacks get it working as libtirpc is Linux specific and isn't provided (and used) by BSDs including MacOS. Free/Open/NetBSD doesn't provide this library at all and that also includes MacPorts and brew(.sh). Since OS specific hacks aren't enouraged in general this probably falls under the "will not fix" category.

ldir-EDB0, 2019-04-23:
That's unfortunate news indeed.  Every other package I've enabled has built ok on macos, admittedly that's not *all* the packages.  I guess a question would be "why does samba4 need a host implementation of libtirpc to build'.  The target version of libtirpc builds fine, as at that stage it's building/cross compiling for a linux system.

diizzyy, 2019-04-23:
As a workaround for now at least unless you need something of fancy stuff Samba4 provides this might of interest? https://github.com/openwrt/openwrt/pull/1788

ldir-EDB0, 2019-04-23:
in fact why are we building samba4 for host as well??

 make[3] -C feeds/packages/libs/libtirpc host-compile
 make[3] -C feeds/packages/net/samba4 host-compile

Andy2244, 2019-04-23:
> in fact why are we building samba4 for host as well??

Mainly for a working `asn1_compile`.

closed 2019-06-28 by ldir-EDB0