openwrt/packages #8812: bridge: check_data_file_clashes: Package bridge wants to install file /usr/sbin/brctl

highjagger, 2019-04-27:
Maintainer: @mar-kolya 
Environment: OpenWrt 18.06.2 r7676-cddd7b4c77

Description:
While trying to install package bridge, following warning occours.

```
Collected errors:
 * check_data_file_clashes: Package bridge wants to install file /usr/sbin/brctl
        But that file is already provided by package  * busybox
 * opkg_install_cmd: Cannot install package bridge.
```

With the usage of `--foce-overwrite` it works. its also needed for further upgrades of busybox.

yousong, 2019-07-04:
I will update here when I got time for this.

The plan is to backport commit openwrt/openwrt@e51b513f and openwrt/openwrt@62be427067ee3883b68bcfb08dfc0c43dce22fa3 to both branch 19.07 and 18.06.  The add alternatives spec for `bridge` and `coreutils` (#9290) package here.

yousong, 2019-07-12:
Should now be fixed for both 18.06 and 19.07

closed 2019-07-12 by yousong