openwrt/packages #8838: rpcbind : missing package in repo. Can't install nfs-kernel-server

mwoz123, 2019-04-29:
Maintainer: @Andy2244 , @tripolar , @ja-pa 
Environment: 
`Linux OpenWrt 4.9.152 #0 Mon Jan 28 08:54:32 2019 mips GNU/Linux,`
tplink wdr-4300,  
`OpenWrt 18.06.2, r7676-cddd7b4c77`

# Description:
I've fresh install of openwrt. Trying to install `nfs-kernel-server` gives me error: missing package **rpcbind**  in repository.

I did updated repository (`~# opkg update`) without any errors & then:
```
root@OpenWrt:~# opkg install nfs-kernel-server
Installing nfs-kernel-server (2.3.3-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/nfs-kernel-server_2.3.3-2_mips_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for nfs-kernel-server:
 *      rpcbind
 * opkg_install_cmd: Cannot install package nfs-kernel-server.
root@OpenWrt:~#
```

I've a clean, unchanged sources.list - should I add any backports for this? 
I'm newbie to openwrt.. 
Please help.

BKPepe, 2019-04-29:
The thing is that rpcbind is not available in OpenWrt 18.06, it is just in master.

BKPepe, 2019-04-29:
Caused by this commit: https://github.com/openwrt/packages/commit/f1ab32ab62c3dfbf14bbe1a683bf43563a141508#diff-03b759f96e4ac1b9707753b91a695fffR40

Fix (most probably) would be this one: https://gitlab.labs.nic.cz/turris/turris-build/commit/8f325bcbe6309d93cb998c0bd86f16088b84b685  but before I send a pull request, I would like to see, if it works. It should as there was in past port map, but who knows.

mwoz123, 2019-04-29:
I assume there's no repo with build packages for master, so I can't easily add repo / or even better download only this package (I've bad memories with mixing repos on linuxes) and install it manually, right? 

Do I understand correctly that I should install packages required for compiling on router, compile manually rpcbind (from master), create package and then install ?

BKPepe, 2019-04-29:
It's a way, how you can achieve it, but we need to fix it to all users.  :)
AFAIK: New packages won't be accepted to 18.06.

BKPepe, 2019-04-30:
rpcbind was added to OpenWrt 18.06.02 by @tripolar in this commit https://github.com/openwrt/packages/commit/7f7fda67bd6656592d486462915966de9e7b65db.

closed 2019-04-30 by BKPepe

mwoz123, 2019-05-03:
Able to install. Thanks :)