openwrt/packages #8839: snort3 fails to build

rakesh2215, 2019-04-29:
Tried building snort3 with latest x86_64 sdk.
Failing to build with below error:

```
  The C++ compiler

    "openwrt_sdk/openwrt-sdk-x86-64_gcc-7.4.0_musl.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-g++"

  is not able to compile a simple test program.

  
    openwrt_sdk/openwrt-sdk-x86-64_gcc-7.4.0_musl.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -ltirpc
```

cshoredaniel, 2019-05-26:
Occurs with ath79 as well.

Strykar, 2019-06-08:
@rakesh2215 Have you tried with current, you can checkout the release you want? I've had trouble building with the SDK in the past. Using snort 2 here.

rakesh2215, 2019-06-08:
Current release 18.6.2 doesn't have snort3.
Snort3 Makefile is added only in snapshot.
Do you mean I use this Makefile and build with 18.6.2?

cshoredaniel, 2019-06-10:
@flyn-org

cshoredaniel, 2019-06-19:
@flyn-org Buildbots see it too: https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/snort3/compile.txt

neheb, 2020-01-22:
That build failure has to do with not having host GCC 32-bit binaries.

Is the libtirpc error still an issue?

DaveRush82, 2020-03-29:
> That build failure has to do with not having host GCC 32-bit binaries.
> 
> Is the libtirpc error still an issue?

Which libtirpc error?

rakesh2215, 2021-01-14:
@DaveRush82     

openwrt_sdk/openwrt-sdk-x86-64_gcc-7.4.0_musl.Linux-x86_64/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/../lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: **cannot find -ltirpc**

flyn-org, 2021-12-15:
I was able to build snort3 using the current openwrt-sdk-bcm47xx-generic_gcc-11.2.0_musl.Linux-x86_64. @rakesh2215, is this still and issue?

rakesh2215, 2021-12-15:
probably not.

hnyman, 2023-01-14:
Currently builds ok for master in buildbot.

closed 2023-01-14 by hnyman