openwrt/packages #8852: strongswan: strongswan-isakmp missing gmp.h

lucize, 2019-04-30:
Maintainer: @stintel 
Environment: master

Description:

building only the meta isakmp fails with

```
OpenWrt-libtool: compile:  i486-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../.. -I../../../../src/libstrongswan -I/home/build/staging/staging_dir/target-i386_pentium4_musl/usr/include -I/home/build/staging/staging_dir/target-i386_pentium4_musl/include -I/home/build/staging/staging_dir/toolchain-i386_pentium4_gcc-7.4.0_musl/usr/include -I/home/build/staging/staging_dir/toolchain-i386_pentium4_gcc-7.4.0_musl/include/fortify -I/home/build/staging/staging_dir/toolchain-i386_pentium4_gcc-7.4.0_musl/include -I/home/build/staging/staging_dir/target-i386_pentium4_musl/usr/lib/libiconv-stub/include -I/home/build/staging/staging_dir/target-i386_pentium4_musl/usr/lib/libintl-stub/include -rdynamic -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/build/staging/build_dir/target-i386_pentium4_musl/strongswan-5.7.2:strongswan-5.7.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/build/staging/staging_dir/target-i386_pentium4_musl/usr/lib/libiconv-stub/include -I/home/build/staging/staging_dir/target-i386_pentium4_musl/usr/lib/libintl-stub/include -include /home/build/staging/build_dir/target-i386_pentium4_musl/strongswan-5.7.2/config.h -MT ../gmp/gmp_diffie_hellman.lo -MD -MP -MF ../gmp/.deps/gmp_diffie_hellman.Tpo -c ../gmp/gmp_diffie_hellman.c  -fPIC -DPIC -o ../gmp/.libs/gmp_diffie_hellman.o
../gmp/gmp_diffie_hellman.c:20:10: fatal error: gmp.h: No such file or directory
 #include <gmp.h>
          ^~~~~~~
compilation terminated.
Makefile:587: recipe for target '../gmp/gmp_diffie_hellman.lo' failed
make[8]: *** [../gmp/gmp_diffie_hellman.lo] Error 1
make[8]: *** Waiting for unfinished jobs....
make[8]: Leaving directory '/home/build/staging/build_dir/target-i386_pentium4_musl/strongswan-5.7.2/src/libstrongswan/plugins/gmpdh'
Makefile:2054: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1

```

`strongswan-mod-gmpdh` is saying  `StrongSwan DH-Groups; no libgmp dep plugin` but it seems that `strongswan-mod-gmp` is still needed

neheb, 2019-07-02:
Is this still an issue?

lucize, 2019-07-02:
yes, I think something is missing in the 305-minimal_dh_plugin.patch

closed 2019-09-16 by lucize