openwrt/packages #8854: strongswan: openssl deprecated

lucize, 2019-04-30:
Maintainer: @stintel 
Environment: master

Description:
compiling openssl module fails to load with

`daemon.info : 00[LIB] plugin 'openssl' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-openssl.so: X509_get_notAfter: symbol not found
` 

openssl says `X509_get_notBefore() and X509_get_notAfter() were deprecated in OpenSSL 1.1.0`

this is a fix

```
--- a/src/libstrongswan/plugins/openssl/openssl_x509.c  2019-04-30 15:30:59.590212295 +0300
+++ b/src/libstrongswan/plugins/openssl/openssl_x509.c  2019-04-30 15:31:32.158213669 +0300
@@ -1137,8 +1137,8 @@
                return FALSE;
        }

-       this->notBefore = openssl_asn1_to_time(X509_get_notBefore(this->x509));
-       this->notAfter = openssl_asn1_to_time(X509_get_notAfter(this->x509));
+       this->notBefore = openssl_asn1_to_time(X509_get0_notBefore(this->x509));
+       this->notAfter = openssl_asn1_to_time(X509_get0_notAfter(this->x509));

        /* while X509_ALGOR_cmp() is declared in the headers of older OpenSSL
         * versions, at least on Ubuntu 14.04 it is not actually defined */
```

@neheb @cotequeiroz 
this should go upstream ?

neheb, 2019-04-30:
Yes of course. openssl_asn1_to_time should also be changed to accept a const parameter. Compatibility defines for OpenSSL 1.0.2 would be a good idea as well.

lucize, 2019-05-07:
will be updated in 5.8.0

closed 2019-05-07 by lucize