openwrt/packages #8888: stubby: digest type is wrong if corr. key contains slashes (uci)

bjmi, 2019-05-03:
Maintainer: @jonathanunderwood 
Environment: OpenWrt 18.06.2

Description:
Some _tls_pubkey_pinset_ contain slashes like this one
`v7rm6OtQQD3x/wbsdHDZjiDg+utMZvnoX3jq3Vi8tGU=`

The corr. resolver config with uci (_/etc/config/stubby_) looks like
```
config resolver
[...]
       list spki 'sha256/v7rm6OtQQD3x/wbsdHDZjiDg+utMZvnoX3jq3Vi8tGU='
```
and will be generated to _/etc/stubby/stubby.yml_
```
      - digest: "sha256/v7rm6OtQQD3x"
        value: v7rm6OtQQD3x/wbsdHDZjiDg+utMZvnoX3jq3Vi8tGU=
```
but determined _digest_ is  wrong. It should be **sha256** only.

https://github.com/openwrt/packages/blob/openwrt-18.06/net/stubby/files/stubby.init#L139 takes the last slash and not the first slash.

jonathanunderwood, 2019-05-09:
Thanks for reporting this. I'll have a think on the right fix: ':' would've been a better choice of separator, but I'd prefer to explore a fix that doesn't break existing working configs.

jonathanunderwood, 2019-05-09:
Have just submitted a PR with a proposed fix: I'd welcome your testing.

jonathanunderwood, 2019-05-10:
Fixed by #8939.

This can be closed.

closed 2019-05-10 by neheb

bjmi, 2019-05-17:
> Have just submitted a PR with a proposed fix: I'd welcome your testing.

stubby - 0.2.6-1 works like a charm. Thank you!