openwrt/packages #8892: python: external feeds and the python build system

pmelange, 2019-05-03:
Maintainer: @commodo 
Environment: architecture independant, python build system
Description: python based packages from external feeds fail to build

Building from an external feed has, as far as I can tell, the following issues:
1. There is not a portable way to include the python[3]-package.mk files
2. Other included python[3]-*.mk files which are included assume that the pacakge is in the packages feed

Part 1)
As stated in commit https://github.com/openwrt/packages/commit/1476739d5f85be70b5db63b439dfe812065b52c7, building python packages from external feeds needs to be addressed.

As recommended by @jefferyto in https://github.com/openwrt/packages/commit/1476739d5f85be70b5db63b439dfe812065b52c7#commitcomment-33384628, including the python[3]-package.mk file using 
```
include $(TOPDIR)/feeds/packages/lang/python/python-package.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk
```
is problematic since it requires that the name of the feed for the https://github.com/openwrt/packages repo be named ```packages``` in feeds.conf.

Part 2)
Assuming that the name is feeds.conf is ```packages``` and the method above is used, then there is still another even bigger problem.  python[3]-package.mk includes python[3]-host.mk using the following code (example taken from python-package.mk):
```
python_mk_path:=$(dir $(lastword $(MAKEFILE_LIST)))
include $(python_mk_path)python-host.mk
```
which causes the makefile to include a file which does not exist.  The result is that the python/host packages are not created and a results in compile error similar to:
```
bash: /home/XY/src/openwrt/staging_dir/hostpkg/bin/python2.7: No such file or directory
```

* Note: If, in the .config, I enable a python package from the packages feeds, then ```make package/python-NAME/compile```then the host packages build.  Following that with ```make``` builds successfully.  But this is not a valid workaround.

pmelange, 2019-05-03:
The feed which I am trying to use is: https://github.com/wlanslovenija/firmware-packages-opkg

I have forked the repo and made the Makefile changes in (feeds.conf format)
https://github.com/pmelange/firmware-packages-opkg.git^89186ef8c4b0346c0bd81a2190e12cadc61bf4b

commodo, 2019-05-06:
The report is a bit confusing.
Because, what you want, seems to be to just to build `tunneldigger-broker`, but then the analysis that is provided has nothing to do with it.

adding
```
include $(TOPDIR)/feeds/packages/lang/python/python-package.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk
```
is the solution to build the packages in  https://github.com/wlanslovenija/firmware-packages-opkg
some other stuff, was also needed to complete the conversion though; mostly the VARIANT stuff ;

I did a quick conversion ( [see this branch, top commit](https://github.com/commodo/firmware-packages-opkg/commits/fix1) ) and it built fine for me.

when you say

> is problematic since it requires that the name of the feed for the https://github.com/openwrt/packages repo be named `packages` in feeds.conf.

well, we only usually care about this repository, and the Python/Python3 in this repository;
i admit, it is not portable, but then, anyone is free to use whatever-else Python/Python3 build from some other repo ;

the old `include_mk` construct [which was somewhat portable] was problematic in some other ways ; mostly outlined [here in this comment](https://github.com/openwrt/packages/commit/1476739d5f85be70b5db63b439dfe812065b52c7#commitcomment-33400278) 

regarding
```
python_mk_path:=$(dir $(lastword $(MAKEFILE_LIST)))
include $(python_mk_path)python-host.mk
```
and

> which (and please correct me if I'm wrong) causes the makefile to include a file which does not exist. The result is that the python/host packages are not created and a results in compile error similar to:

i am a bit confused as to how that happens;
this seems to have worked fine up until now, so i am guessing there is something weird/custom on your system ;

hopefully you are trying out a more recent version of OpenWrt (at least 18.06) with this stuff;

pmelange, 2019-05-06:
@commodo Thank you for spending the time to look at the issue.  And extra thanks for taking a look and working on the Makefiles.

> adding
> 
> ```
> include $(TOPDIR)/feeds/packages/lang/python/python-package.mk
> include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk
> ```
> 
> is the solution to build the packages

I would recommend adding this to lang/python/README.md as an extra note for external feeds (with the caveat that it is not portable and it is expected that the packages feed is called ```packages```). 

This is the original reason for this issue.

----
But to my specific issue, which again I am very appreciative that you have given your time and energy to this already...

My build environment is on a Debian 9.8 system, with all required build environment packages.  I'm using the SDK from snapshot (because of hardware support and in anticipation of 19.x).

My feeds.conf is:
```
src-git base https://git.openwrt.org/openwrt/openwrt.git
src-git packages https://git.openwrt.org/feed/packages.git
src-git luci https://git.openwrt.org/project/luci.git
src-git routing https://git.openwrt.org/feed/routing.git
src-git nodewatcher https://github.com/pmelange/firmware-packages-opkg.git;19x_Makefile_updates
```
I now have working Makefiles for python2 and python3 (even though the tunneldigger-broker code isn't ready yet).  You can take a look at https://github.com/pmelange/firmware-packages-opkg/commits/19x_Makefile_updates to see them.

commodo, 2019-05-07:
> I would recommend adding this to lang/python/README.md as an extra note for external feeds (with the caveat that it is not portable and it is expected that the packages feed is called `packages`).

will do that;

code looks good for those packages

commodo, 2019-05-07:
I forgot to ask: was there a question related to the code ? or is there an issue you are stuck with ?
[ it wasn't clear to me, if there is ]

pmelange, 2019-05-07:
I think I can work out any the issues that are left (for example, rewriting the init script for tunneldigger-broker).  I think the most time consuming (aside from compiling) was trying to find an example of an external feed which uses the up-to-date python build system.  

Now that the instructions are going into README.md, then in the future this won't be a problem.

Should I close this issue now, or wait for a merged commit with the changes to README.md?

commodo, 2019-05-07:
> Should I close this issue now, or wait for a merged commit with the changes to README.md?

Your choice;
I usually keep email as a personal task-tracker.

pmelange, 2019-05-07:
Oh, one thing I just thought of.  In the old Makefile, there was ```PKG_BUILD_DEPENDS```, which I do not have in the new Makefile.  It compiles just fine though.  

Should something like the following be added?  Is it really necessary?
```
HOST_PYTHON_PACKAGE_BUILD_DEPENDS:="cffi==$(PKG_VERSION)"
HOST_PYTHON3_PACKAGE_BUILD_DEPENDS:="cffi==$(PKG_VERSION)"
```

commodo, 2019-05-07:
> Should something like the following be added? Is it really necessary?
> 
> ```
> HOST_PYTHON_PACKAGE_BUILD_DEPENDS:="cffi==$(PKG_VERSION)"
> HOST_PYTHON3_PACKAGE_BUILD_DEPENDS:="cffi==$(PKG_VERSION)"
> ```

Some packages (like cffi) need to be installed on the host-Python first, and only then they can cross-compile.
It's also very difficult to patch them in a way that they don't need presence in host-Python.
Or, they need some tool from the package.
For those package, the mechanism (which isn't great) is the one described above.

Fortunately there aren't many of these packages.

The packages in firmware-packages-opkg don't need this mechanism.
So, these aren't necessary for these packages.