openwrt/packages #8893: Python 2: Transition programs to Python 3

jefferyto, 2019-05-03:
This is a meta-issue to track progress for transitioning programs that use Python 2 to Python 3.

These should be packages that use Python in some way, as part of their core functionality. Packages that provide a Python 2-only library or plugin will be tracked in a separate issue (TBD).

| Section | Package | Status | Maintainers | Related issues/PRs |
| --- | --- | --- | --- | --- |
| lang | node | | @blogic @ianchi | |
| lang | node-mozilla-iot-gateway | | @ratkaj | |
| mail | mailman | | @Shulyaka | |
| net | aircrack-ng | | @ZeroChaos- | |
| net | danish | | @smutt | |
| net | nmap | | @nunojpg | |
| net | obfsproxy | | @jefferyto | |
| net | openvswitch | Done | @yousong | #8797 |
| net | radicale | | | |
| net | samba4 | | @Andy2244 | |
| net | scapy | Done | @MikePetullo | #8824 |
| net | seafile-ccnet | | @commodo | |
| net | seafile-seahub | | @commodo | |
| net | seafile-server | | @commodo | |
| net | sysrepo | | @mislavn | |
| sound | upmpdcli | | @ignisf | |
| utils | bluez | | | |
| utils | gammu | | @aTanW | |
| utils | hplip | | @luizluca | |
| utils | yunbridge | | @blogic | |

Maintainers: If your program provides a Python 2 library package, please be reminded to keep it around (if possible) until the mass removal in 2020.

Please comment on / reference this issue as things develop. I will keep this post updated "periodically" :joy:

neheb, 2019-05-03:
Nice! Want this pinned?

neheb, 2019-05-03:
Maintainer of aircrack-ng is @jbenden

jefferyto, 2019-05-03:
@neheb 

> Nice! Want this pinned?

Sure :laughing: 

> Maintainer of aircrack-ng is @jbenden

Updated - thanks!

jefferyto, 2019-05-03:
Added devel/asu (ping @aparcar) because while the server itself uses Python 3, it also wants to setup an imagebuilder environment on target and so is pulling in Python 2. (This is mostly speculation though, I haven't looked too deeply into the details.)

If so, sounds like asu won't be resolved until OpenWrt core has moved to Python 3.

smutt, 2019-05-04:
I'm currently rewriting Danish in Rust so there will be no Python 3 version.
<https://github.com/smutt/danish-rust>

I hope to be done with it by December and I'll call it Danish 2.0. The Python version will be abandoned and only the Rust version will be supported going forward.

The current Python 2 version worked fine for its purposes, and I've been running it for over a year on my home OpenWRT box. But it was an experiment in doing middle-box DANE for HTTPS that I now want to get more serious about and also bring to a more general Linux audience, not just OpenWRT. And for that I needed to re-implement in something like Rust or C/C++.

jefferyto, 2019-05-04:
@smutt No problem - thanks for the update :smile:

Andy2244, 2019-05-14:
Samba4.10.x will only fully support python3 from here on out, so if i get over this issue #8944 we will be fine.

jefferyto, 2019-05-14:
@Andy2244 Thanks for the update!

aparcar, 2019-05-14:
`asu` is python3 only, @dangowrt did the porting for various dependencies, however only Python3 if I recall correctly...

jefferyto, 2019-05-14:
@aparcar So you're saying the `+python-light` in [these lines](https://github.com/openwrt/packages/blob/4920a842f4a6939b1254a6247fe1736308e0ef86/devel/asu/Makefile#L25-L30) (labelled `IMAGEBUILDER_DEPENDS`) should be Python 3 and not Python 2?

aparcar, 2019-05-14:
Good point, I'm not sure if the ImageBuilder needs python2 or can handle also 3, the [docs](https://openwrt.org/docs/guide-user/additional-software/imagebuilder) require `python`...

@dangowrt do you know if the ImageBuilder itself is already Python3 compatible? If not, that's kind of a new issue...

cshoredaniel, 2019-05-26:
radicale 1.x is never going to Python3 -- radicale2 replaces radicale package and requires Python 3 -- the reason for renaming and having both in tree was to give current radicale users a chance to transition to radicale2 (requires a database conversion, which I think has to be done from radicale 1.x later version).

jefferyto, 2019-05-26:
@cshoredaniel Thanks for the update - I guess I have two questions:

1. Is there any reason why a radicale 1.x user would want to stick with 1.x and not upgrade to 2.x?

2. Is there a way we can prompt 1.x users to upgrade to 2.x, before the 1.x package is gone (and they potentially lose the ability to convert their database)? Maybe log a warning to syslog on every startup? :laughing:

jefferyto, 2019-05-26:
@cshoredaniel Wait, sorry I just checked the radicale package and there are Py2 and Py3 versions of the package?

If this is true, we can just remove the Py2 version and keep the Py3 version... But I guess the question still is, should we keep 1.x around?

ignisf, 2019-05-26:
upmpdcli's songcast web interface should be compatible with python3. Will take a look at it and update.

cshoredaniel, 2019-05-26:
@jefferyto Oh, you're right, I forgot there is a Py3 radicale, just not a Py2 radicale2...I think we should keep radicale 1.x for 19.0x but have a deprecation alert (actually there might already be such -- it's been a while since I last looked and I might be confusing with another distro).

jefferyto, 2019-05-27:
@ignisf Thanks for the update!

luizluca, 2019-06-25:
For hplip, it should be compatible with python3 since 3.15.2. I'll test, commit and report.

luizluca, 2019-06-26:
@jefferyto , hplip is clean. It didn't actually need python for OpenWrt usage.

jefferyto, 2019-06-29:
@luizluca thanks for the update!

jefferyto, 2019-07-02:
I [asked](https://trac.torproject.org/projects/tor/ticket/31057) the folks at Tor about obfsproxy. It's not actively maintained anymore and obfs4proxy is the official successor, so I think this will be Won't Fix for obfsproxy.

Andy2244, 2019-07-10:
Done for samba 4.10+ via #9440.

jefferyto, 2019-07-10:
@Andy2244 thanks for the update!

Andy2244, 2019-07-21:
Had to revert to the samba 4.9 series, will try python3 with 4.11 again, so still open.

MartB, 2019-07-31:
#9579 also affects libtalloc which is a compile breaker

neheb, 2019-08-13:
upmpdcli is almost done: https://github.com/openwrt/packages/pull/9687

neheb, 2019-08-13:
node-mozilla-iot-gateway was removed from the tree.

neheb, 2019-08-13:
@nxhack deals with node nowadays.

jefferyto, 2019-08-13:
@neheb Thanks for the updates :+1:

nxhack, 2019-08-15:
@neheb 
The latest version of node.js(v12.x) is supporting python3.

But the v12.x series is still not stable, big endian support has been dropped, and cross-build support has been delayed, making it difficult to port to embedded systems.

It will take a little more time.

jefferyto, 2019-08-16:
@nxhack Thanks for the update - v12 is [scheduled](https://nodejs.org/en/about/releases/) to enter LTS status in October, do you think these issues can/will be resolved by then?

nxhack, 2019-08-19:
@jefferyto 
I'm working hard ;) on porting. However, MIPS version (including little endian) may disappear.

neheb, 2019-08-19:
Do MIPS devices even have enough RAM to run node?

nxhack, 2019-08-19:
it works but slow with extroot + swap.
without hardware fpu, super slow.

but it works excellent on armv6 with fpu.

jefferyto, 2019-08-27:
@cshoredaniel Thanks for the update - I wasn't even aware that uwsgi-cgi relied on Python :joy:

nunojpg, 2019-10-06:
Hi, I currently don't have a device with enough flash to run Python. Could someone test the Ndiff updated to Python3 from #10162?

aTanW, 2019-10-07:
gammu package does not depend on python.

neheb, 2019-10-07:
It has a conditional dependency on it.

From my reading of the git log, it seems it supports Python 3: https://github.com/gammu/gammu/search?o=desc&q=python&s=committer-date&type=Commits

neheb, 2019-10-07:
Based on my reading of the aircrack-ng code, it doesn't care which version of python is used?

https://github.com/aircrack-ng/aircrack-ng/blob/32a9915b6873a3bcf2bdbb7116fd9fe01ab44154/build/m4/aircrack_ng_ext_scripts.m4#L44

neheb, 2019-10-07:
Very unlikely that YunBridge will be ported: https://github.com/arduino/YunBridge/issues/31

aTanW, 2019-10-07:
@neheb It has no python code in source tree. Docs may lag. Python bindings in separate package python-gammu in PyPI.

jbenden, 2019-10-08:
> Based on my reading of the aircrack-ng code, it doesn't care which version of python is used?
> 
> https://github.com/aircrack-ng/aircrack-ng/blob/32a9915b6873a3bcf2bdbb7116fd9fe01ab44154/build/m4/aircrack_ng_ext_scripts.m4#L44

The build scripts will accept any python version, and the setup.py might handle any version too. But a quick check shows that the actual tooling does not pass a py3 syntax check. I do think a PR was sent in that fixes the py3 syntax errors, though...

-Joe

neheb, 2019-10-09:
@jebnden on a sidenote, that procps-ng dependency is partially bogus. I posted a patch that replaces ps + grep with only pgrep (BusyBox ps is missing functionality. pgrep works fine) but no progress has been made on that end: https://github.com/aircrack-ng/aircrack-ng/pull/2066

commodo, 2019-10-09:
> test

I forgot to mention this earlier.
I also don't have many routers at-hand to test stuff.
I usually spin-up an x86 VM and test some packages; admittedly some packages [like nmap] do require more setup than others, but some basic testing is doable via VM.

Ansuel, 2019-10-11:
@jefferyto uwsgi port is done

jefferyto, 2019-10-11:
@Ansuel thanks for the heads-up :+1:

jefferyto, 2019-11-04:
@aparcar @nxhack @Shulyaka @jbenden @nunojpg @Andy2244 @commodo @mislavn @blogic:

Just a gentle reminder to update your package(s) from Python 2 to 3 (or to indicate your package(s)'s "Won't Fix" status). (If you are already working on this, then by all means please continue.) As you may know, Python 2 will reach end-of-life in 2 months.

(After these program packages are updated/resolved, we will also need to sort through all Python 2 library packages.)

neheb, 2019-11-04:
I'll handle sysrepo here: https://github.com/openwrt/packages/pull/9491

Upstream has support for both versions.

jbenden, 2019-11-05:
aircrack-ng's master repo now supports Py3. There is no planned release date, yet.

commodo, 2019-11-05:
@jefferyto 
Apologies for the silence on this.

Regarding seafile, that is a bit on-hold.
The suite is a bit of a monster [in terms of Python code-size], and they haven't done any release with Python3; there is some code converting it to Python3, but it's still in progress. I tried to update it to Python3 myself, but that was more work than I like [I wasted 3-4 days trying to convert it to Python3].

seafile is bit of a pain for me, also because of Django 1; while Django LTS is currently at version 2, they still use Django 1. At least I found the elegant solution to split django into 2 packages [ django1 & django] and limit `django1` to Python2, and make `django` Python3 only.

So, for now, what's done, is that all seafile deps both support Python 2 & 3.
I'll see about maybe trying to rework the entire Seafile package to be easier to maintain; this is mostly with the intent of converting it to Python3.
Hopefully they'll take my PRs; I opened one a few months ago and it's still open with no comments.

jefferyto, 2019-11-05:
@neheb Thanks for the update!

@jbenden I was going to ask if it's possible to backport the changes, but I see their last release was from last December and master is more than 500 commits ahead... Would packaging from master be an option?

@commodo I see they have merged PRs for Python 3 (haiwen/seafile-server#238, haiwen/seahub#4076, haiwen/ccnet-server#35) but they are all for master, while their current releases (I think) are from the 7.0 branches. Their [roadmap](https://www.seafile.com/en/roadmap/) has the Python 3 migration for 7.1 in December, so perhaps we will have to wait and see.

For haiwen/seahub#3607, they may not be concerned with cross-compiling, since they distribute their own binaries. It may be a patch we carry ourselves.

commodo, 2019-11-06:
> @commodo I see they have merged PRs for Python 3 ([haiwen/seafile-server#238](https://github.com/haiwen/seafile-server/pull/238), [haiwen/seahub#4076](https://github.com/haiwen/seahub/pull/4076), [...]

yep; i was also thinking about waiting for a new release [which includes Python3 support]; i wanted to do an upgrade to 7.0 [a few weeks ago], but did not see the benefit it if I can't switch to Python3;

well, the thing is i tried to see about maybe backporting Python3 to 7.0 and got annoyed, so, decided not to do it just yet; 

hopefully i'll have some time to clean the packaging a bit;
the way it's packaged in OpenWrt now, is a bit better than before, but I feel I would like to re-visit it; it feels like the whole thing [packaging seafile] was rushed when it was added;

neheb, 2019-11-07:
sysrepo should be good to go.

jbenden, 2019-11-15:
@jefferyto The changes should not be back-ported, yes. There has been talk of pushing out a new release for months, but hasn't gone anywhere - yet. If possible, I think it should be packaged from a known working commit, directly from GitHub. Thoughts?

jefferyto, 2019-11-15:
@jbenden Sounds good - I guess the question is how to find a commit that is known good.

tiagogaspar8, 2019-11-22:
OK, just an update, I saw from the makefile that asu is already suing python3.
My question is: when january comes will the python package removed? If so what happens t the other packages?

jefferyto, 2019-11-22:
@tiagogaspar8 

>OK, just an update, I saw from the makefile that asu is already suing python3.

asu still depends on [python-light](https://github.com/openwrt/packages/blob/9d57b11e445d34fa2604b57fb1a1d075404acf8e/devel/asu/Makefile#L26).

> Also, shouldn't mailman be updated to version 3?

That depends on the maintainer (@Shulyaka) or someone else contributing the update.

> My question is: when january comes will the python package removed? If so what happens t the other packages?

Please see https://github.com/openwrt/packages/tree/master/lang/python#python-2-end-of-life.

neheb, 2019-11-22:
According to: https://gitlab.com/mailman/mailman/blob/master/src/mailman/docs/install.rst , Python 3 is supported and Python 2 is not.

tiagogaspar8, 2019-11-26:
> asu still depends on python-light.
You're right! Sorry.

> That depends on the maintainer (@Shulyaka) or someone else contributing the update.
I'd do it but I have no idea how.

> edit: Ah I see. Mailman needs to be transitioned to version 3.
Yeah ahahaha I hope someone can do it until january.

Andy2244, 2019-11-29:
Quick update regarding samba4, there is now a official patch that solves the main issue that prevented me to switch to the 4.11 branch aka python3. I think i can work around, provide my own patches for the remaining issues.

[19 more replies]

closed 2020-04-03 by jefferyto