openwrt/packages #8903: shadowsocks-libev: how to config ss_server using plugins?

jackcctse, 2019-05-05:
Maintainer: @yousong 
Environment: (MediaTek MT7621, YouHua WR1200JS, OpenWrt SNAPSHOT r9782-09fe18f042)

Description:
I am not sure if this is a bug or configuration issue. ss_server works but I'm trying to config it with plugins, i.e. obfs-server. With below configuration it doesn't work / not connected at all. Can you let me know what is the correct way to using plugins?

```
config ss_server
	option server_port '9001'
	option mode 'tcp_and_udp'
	option server '0.0.0.0'
	option method 'chacha20-ietf-poly1305'
	option password '******'
	option plugin '/usr/bin/obfs-server'
	option plugin_opts 'obfs=tls'
```

yousong, 2019-05-05:
Hi @jackcctse , plugin options are not supported yet.  Please try patching `/etc/init.d/shadowsocks-libev` like the following and see if it works for you.  The option name should suit your posted configuration just fine.

```
From f85eb0bf922e8cdf06ce80fe36605d33d2b73b30 Mon Sep 17 00:00:00 2001
From: Yousong Zhou <yszhou4tech@gmail.com>
Date: Sun, 5 May 2019 16:53:39 +0000
Subject: [PATCH] shadowsocks-libev: add plugin options support

Reference: https://github.com/openwrt/packages/issues/8903

---
 net/shadowsocks-libev/Makefile                     | 2 +-
 net/shadowsocks-libev/files/shadowsocks-libev.init | 6 +++++-
 2 files changed, 6 insertions(+), 2 deletions(-)

diff --git a/net/shadowsocks-libev/Makefile b/net/shadowsocks-libev/Makefile
index 3214a3cc5..b1dc6bfe7 100644
--- a/net/shadowsocks-libev/Makefile
+++ b/net/shadowsocks-libev/Makefile
@@ -14,7 +14,7 @@ include $(TOPDIR)/rules.mk
 #
 PKG_NAME:=shadowsocks-libev
 PKG_VERSION:=3.2.5
-PKG_RELEASE:=1
+PKG_RELEASE:=2
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
 PKG_SOURCE_URL:=https://github.com/shadowsocks/shadowsocks-libev/releases/download/v$(PKG_VERSION)
diff --git a/net/shadowsocks-libev/files/shadowsocks-libev.init b/net/shadowsocks-libev/files/shadowsocks-libev.init
index 42ffe9e0d..2f5168177 100644
--- a/net/shadowsocks-libev/files/shadowsocks-libev.init
+++ b/net/shadowsocks-libev/files/shadowsocks-libev.init
@@ -1,6 +1,6 @@
 #!/bin/sh /etc/rc.common
 #
-# Copyright (C) 2017 Yousong Zhou <yszhou4tech@gmail.com>
+# Copyright (C) 2017-2019 Yousong Zhou <yszhou4tech@gmail.com>
 #
 # This is free software, licensed under the GNU General Public License v3.
 # See /LICENSE for more information.
@@ -73,6 +73,8 @@ ss_xxx() {
 	[ -z "$mode" ] || json_add_string mode "$mode"
 	[ -z "$mtu" ] || json_add_int mtu "$mtu"
 	[ -z "$timeout" ] || json_add_int timeout "$timeout"
+	[ -z "$plugin" ] || json_add_string plugin "$plugin"
+	[ -z "$plugin_opts" ] || json_add_string plugin_opts "$plugin_opts"
 	[ -z "$user" ] || json_add_string user "$user"
 	json_dump -i >"$confjson"
 
@@ -237,6 +239,8 @@ validate_common_options_() {
 		'mode:or("tcp_only", "udp_only", "tcp_and_udp"):tcp_only' \
 		'mtu:uinteger' \
 		'timeout:uinteger' \
+		'plugin:string' \
+		'plugin_opts:string' \
 		'user:string'
 }
 
```

jackcctse, 2019-05-06:
Amazing! Added these 4 lines and ss-server flies. is this patch going to be included in Master branch? 

This has been annoying me sometime. Thank you very much!

closed 2019-05-06 by jackcctse

yousong, 2019-05-06:
Good to know that.  The change was just pushed to both master branch here and LuCI repo.

jackcctse, 2019-05-06:
That will be great. can you make LuCI -> Remote Servers available with plugin options as well? I think it makes sense to have it. At the moment I am direct editing the config file to add plugins option, it works but always handy to get all setting done from LuCI.

yousong, 2019-05-06:
I see.  The option should better be added as part of the `server` section instead of to each individual component.

Servers tab in LuCI app will need to rewritten though.  Currently it's a table already with too many columns crammed in.  Sorry, I do not have time for this these days.   Time budget is a bit tight these day

jackcctse, 2019-05-07:
I understand. Alright I can live with direct editing, as least it works.

Thank you for what you've done.

yousong, 2019-05-09:
Hi @jackcctse 

I am writing to let you know that changes has been made that `plugin` and `plugin_opts` now need to be specified as options of `server` section.  This applies to `ss_server` section as well.  The LuCI app was also changed in accordance.

Thank you for the issue report and suggestions.

jackcctse, 2019-05-09:
Thanks a lot @yousong. I synced latest source and compiled snapshot r9947-165d598521. Everything looks fine but seems there's a bug on LuCI (at Remote Server tab). I put a screenshot below to explain the issue.

https://imgur.com/AooW0L0

yousong, 2019-05-09:
@jackcctse , it's a known issue openwrt/luci#2624.   Sorry that it has been there for a while.  Hopefully I can find some time to tackle it ;)

jackcctse, 2019-05-10:
I understand. It doesn't harm at all, take your time :)