openwrt/packages #8904: opennhrp: depends on ipsec-tools, which is being removed

nmeyerhans, 2019-05-05:
Maintainer: @DerArtem 
Environment: any

Description:

As documented in #7832, ipsec-tools is being removed from the packages repository. Because opennhrp depends on ipsec-tools, it will be broken by this change. I am not familiar with opennhrp, so I don't know what components of ipsec-tools it uses, or if it already supports alternatives, but it would be advisable to look into this. If it depends on racoon for IKE negotiations, consider migrating to [strongswan](/openwrt/packages/net/strongswan) or [libreswan](/openwrt/packages/net/libreswan). If it just depends on the setkey tool, then `ip xfrm` should be a suitable replacement.

lucize, 2019-05-06:
this https://github.com/openwrt-routing/packages/pull/319 claims to have nhrp support, didn't test it

nmeyerhans, 2019-09-01:
opennhrp was removed with #9881

closed 2019-09-01 by nmeyerhans